Field-Robotics-Lab/dave #4: Neutral bouyancy

bsb808, 2020-03-13:
The UUV Simulator [HydrodynamicModel Plugin](https://github.com/uuvsimulator/uuv_simulator/blob/e44aba6493bbbe55024c39286cc9b9e1d966e409/uuv_gazebo_plugins/uuv_gazebo_plugins/src/HydrodynamicModel.cc)  parses the `neutrally_buoyant` SDF parameter and calls a corresponding method (see https://uuvsimulator.github.io/packages/uuv_simulator/docs/api/gazebo::BuoyantObject/#public-voidsetneutrallybuoyant).

Not much documentation, but the name seems to suggest that this would setup the plugin so that the ROV was stable at a fixed alt/depth.  

However, setting this parameter to `1` in rexrov xacro description (https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_descriptions/urdf/rexrov.gazebo.xacro), which is called by the example `roslaunch uuv_dave uuv_dave.launch` does not seem to have the desired effect.

bsb808, 2020-03-13:
Also see https://github.com/uuvsimulator/uuv_simulator/issues/77

woensug-choi, 2020-09-02:
This is related to https://github.com/Field-Robotics-Lab/glider_hybrid_whoi/issues/8 .

bsb808, 2020-10-08:
Done

closed 2020-10-08 by bsb808