Field-Robotics-Lab/dave #12: Develop feedback control functionality

bsb808, 2020-05-13:
Goal is to have station-keeping functionality demonstrated (examples and tutorial/demo instructions).

Typically for ROVs the motion control is decomposed into independent DOFs:

* Auto-heading (yaw control)
* Auto-depth 
* Auto-altitude
* Auto-xy (station keeping)

Each of these DOFs can be enabled/disabled independently.  For our work this would be actuated via joystick or keyboard teleop.

## Previous work

* UUV Simulator - [DP Tutorial](https://uuvsimulator.github.io/packages/uuv_simulator/docs/tutorials/creating_new_dp_controller/) .   Dynamic Positioning is synonymous with station-keeping.  I have not dug into this process, but at first glance it appears to be overly complicated.

## Approach

 * Teleoperator engages each feedback control for each DOF independently 
 * If disengaged (open-loop), the teleoperation commands provide a thrust command to the vehicle.  this passes through the [thrust manager](https://uuvsimulator.github.io/packages/uuv_simulator/docs/tutorials/config_thruster_manager/) so that the thruster allocation matrix allows the linear/angular thrust command to be decomposed into commands for the individual thrusters.
  * If engaged (closed-loop)
    * If the command is zero (within deadband of joystick) either
      * The velocity setup is zero (this is a bit easier) OR
      * The vehicle goes into position control mode
    * If the command is non-zero, the teleoperation command maps to a velocity setupoint.

## Where to start

Start with auto-heading and/or auto-depth.  Doing a single degree of freedom at a time is quite a bit easier.  Auto-xy can be a little tricky.

M1chaelM, 2020-09-15:
I think this is resolved by using the `joy_velocity.launch` file instead of `joy_accel.launch` under `uuv_control_cascaded_pid` and is now the default for our `uuv_dave.launch` demo. @bsb808 does this resolve the issue?

bsb808, 2020-09-24:
Resolved by the closed loop examples in the tutorials where ROV uses thrusters and sensor feedback for station keeping

bsb808, 2020-09-24:
Resolved by the closed loop examples in the tutorials where ROF uses thrusters and sensor feedback for station keeping

closed 2020-09-24 by bsb808