Field-Robotics-Lab/dave #13: Static vehicle option

bsb808, 2020-05-13:
Doing coordinated, whole-body, mobile manipulation is challenging.  For many examples it is best to start with a simpler example where the vehicle is static - approximating a scenario where the vehicle is on the bottom or held stationary by a perfect feedlback control system.

Develop a convenient interface and example such that the vehicle can be set to static when spawned in the world to facilitate testing the manipulation independent of the vehicle motion control.  Ideally this would be a command line argument when launching the scenario.

Youssef-Khaky, 2020-05-13:
@bsb808 I think there is an example like that. It spawns the UUV and fixes it in place in front of a bop board.
`roslaunch uuv_gazebo rexrov_bop_panel.launch`

Youssef-Khaky, 2020-05-13:
[image]

It looks like this. Apart from getting lost in the fog when adjusting the camera angles and the very dark bop panel, I think it demonstrates the manipulation functionality well.

M1chaelM, 2020-09-15:
@bsb808 Does @Youssef-Khaky's comment resolve this issue?

bsb808, 2020-09-24:
Resolved by the closed loop examples in the tutorials where ROF uses thrusters and sensor feedback for station keeping

closed 2020-09-24 by bsb808