Field-Robotics-Lab/dave #18: Verification of ft sensing for manipulator force feedback.

bsb808, 2020-05-26:
Testing PR#16 raises some questions about the validity of the torque sensor plugin outputs and their use for force feedback.  See the video: https://vimeo.com/422977387  There seem to be some simulation artifacts.

Can we test a subset of joints by adding external forces to gain confidence that they are providing trustworthy results?  

A suggested step is to simplify the example from the PR to a standalone manipulator on a fixed pedestal with no ocean world (just empty gazebo world).  This would simplify the testing and allow us to do some sniff tests of the outputs.

bsb808, 2021-02-01:
At this point, this is not on our list of active efforts.  We are leaving the issue open, but not actively working on it at this time.

crvogt, 2021-12-09:
Closing, see #167

closed 2021-12-09 by crvogt