Field-Robotics-Lab/dave #25: Remove the dave/sonar_dave_plugins directory

BruceMty, 2020-06-11:
The dave/sonar_dave_plugins directory contains a copy of the uuv image_sonar_ros_plugin Sonar model.  It was to contain a custom Sonar model.  Now sonar models are managed elsewhere.  This Sonar model copy provides libimage_sonar_dave_plugin.so.  The only entry to this plugin is sonar_dave_uuv/urdf/sonar_snippets.xacro and its only entry is dave/uuv_dave/urdf/sensor_snippets.xacro.  Its significant use is to provide Sonar for the primary example:

    roslaunch uuv_dave uuv_dave.launch

The fix must include redirection to UUV's libimage_sonar_ros_plugin.so and to .xacro files so that this example still works.  We may wish to perform this fix when a new Sonar model becomes available.

bsb808, 2020-09-09:
Could we use the sonar plugin from uuv_simulator in the example (uuv_dave.launch) and then remove the copy of the image sonar pluign in this directory?

BruceMty, 2020-09-24:
@bbingham-nps @M1chaelM this issue may be closed when https://github.com/Field-Robotics-Lab/dave/pull/64 is merged.

bsb808, 2020-09-24:
@BruceMty Roger - please close the issue when you merge your PR.

closed 2020-09-25 by BruceMty