Field-Robotics-Lab/dave #27: OpenCV compilation error

PenguineZhang, 2020-07-23:
Is there opencv version requirement to compile the code? My workaround is to change CV_AA to CV_L1 as suggested by the compiler. My machine is using OpenCV 4.1.0. See attached and let me know what you think. 
[image]

bsb808, 2020-07-23:
This does seem like a dependency issue with the setup.  That source file does make use of...
```
#include <cv_bridge/cv_bridge.h>
#include <opencv2/core/core.hpp>
```

Can you make sure you have the setup as described [here](https://github.com/Field-Robotics-Lab/dave/wiki/Install-Directly-on-Host).  One thing that causes issues is to make sure you have the updated Gazebo 9 from OSRF.  It should be `Gazebo multi-robot simulator, version 9.13.1`

@BruceMty Do we need the sonar_dave_plugins anymore?  It look like this is a copy of the source file from the [uuv_simulator](
https://github.com/uuvsimulator/uuv_simulator/blob/master/uuv_sensor_plugins/uuv_sensor_ros_plugins/src/gazebo_ros_image_sonar.cpp)  Can we remove that part of the repository?

BruceMty, 2020-07-23:
@bsb808 , this is a copy but we currently need it, see https://github.com/Field-Robotics-Lab/dave/issues/25.  I expect that uuv_simulator will raise the same CV complaint.  For me, typing "pkg-config --modversion opencv" indicates CV version 3.2.0.

PenguineZhang, 2020-07-23:
@BruceMty @bsb808 Upgrading to Gazebo 9.13.2 (should be the latest) still raised the same compilation issue. The output of "pkg-config --modversion opencv" also indicates 3.2.0. It turned out CV_AA is defined in "opencv2/imgproc/imgproc_c.h". When I added include <opencv2/imgproc/imgproc_c.h>, it compiled sucdessfully.

bsb808, 2020-07-23:
That is strange that we aren't seeing the same thing - suggests that the development environment is not equivalent.  This is a perfect reason why we need some CI/CD to provide checks for this in a clean dev. env. 

I suggest we incur the technical debt and use this as our first test case of @PenguineZhang submitting a PR through gitlab that eventually makes it to the github repo.

PenguineZhang, 2020-07-23:
@bsb808 Is there a mirror repo already at Gitlab? I have not received the invite to Gitlab yet.

M1chaelM, 2020-07-23:
@PenguineZhang I just got it set up and sent you an invite. Let me know if you have any trouble.

mabelzhang, 2020-09-25:
First, adding the `#include` seems the right thing to do anyway. I suspect a strict linter would tell you to include the header files you're using. As you mentioned, `CV_AA` is defined in `imgproc_c.h` even in 3.2.0: https://docs.opencv.org/3.2.0/globals_c.html

Second, for the sake of looking into the issue, I have 3.2.0 installed from binaries in a homemade Docker image, and I don't have compiler errors. Bruce's command above gives me 3.2.0 as well.

Do you have two versions of OpenCV installed? 4.1.0 from source (as I think you mentioned), and 3.2.0 from binaries (along with other things like ROS that might have installed it). It could be that pkg-config picked up your binary install, and the compiler picked up the source install and produced the error.

What if you add the following line in your `dave/sonar_dave_plugins/sonar_dave_uuv/CMakeLists.txt` after `find_package(OpenCV REQUIRED)`,
```
message("OpenCV version: " ${OpenCV_VERSION})
```
and see what it says at compile time? That would tell us the version of OpenCV that CMake is getting and might give us some clues.

Third, what is the command you ran to compile? Usually, this kind of discrepancy can happen if that header file is indirectly included by some other file being compiled in the collection - I'm compiling everything in the workspace, so one of the files could have included a file that includes `imgproc_c.h`. If you're compiling only a few packages, maybe none of them happened to do that, and you'd get that error. In that sense, you are compiling in a cleaner environment than I am, and that's usually how CI machines are set up, so that they're stricter on missing dependencies.

However, that doesn't seem the case here. I `rm -rf build install` and ran `colcon build --packages-up-to sonar_dave_uuv` (this compiles `sonar_dave_uuv` and all its dependency packages, and everything still compiled without error. The packages that this command compiled are `sonar_dave_uuv`, `uuv_gazebo_plugins`, and `uuv_sensor_ros_plugins_msgs`. So I think something else is going on.

mabelzhang, 2021-02-01:
Closing as issue seems to have been resolved. Feel free to reopen if it crops up again.

closed 2021-02-01 by mabelzhang