Field-Robotics-Lab/dave #34: docker does not support as many configuration options as we offer

M1chaelM, 2020-08-13:
The Docker scripts we have now were quickly ported over from our previous VRX environment. They offer command line options that allow the user to specify:
1. whether the base image is Nvidia-enabled or not;
1. whether the base image runs Ubuntu 18 with ROS melodic and Gazebo 9 or Ubuntu 16 with ROS kinetic and Gazebo 7

The combination of these choices allow the user to build 4 different varieties of images.

Problem:
* Of these options, the only one that we are actively testing Ubuntu 18.04/Melodic/Gazebo9 with an Nvidia GPU.

Fix:
* I propose to drop support for everything except the configuration we're actually using.

M1chaelM, 2020-08-13:
@bsb808 and @caguero I added you to this issue to get your opinion. If you are OK with the proposed fix I will implement it. I'm mostly wondering whether there is any reason we need to support a non-nvidia Docker image.

M1chaelM, 2020-08-14:
I just tested our non-Nvidia Docker image and it runs (as long as the host is not using the Nvidia driver), but gives a realtime factor of around .35. The only reason I can think of to actively support this is if we are required to have a non-GPU image for our github actions testing pipeline. @caguero Do you know whether there is any problem using the nvidia-docker image for testing in Github?

caguero, 2020-08-17:
My suggestion is to support two flavors of Ubuntu 18.04/Melodic/Gazebo9, one with Nvidia and the second with non-Nvidia support. We know that this setup works with GitHub actions, as we're using the non-nvidia image for CI. Once that's in place, we could do more experiments and test if the Nvidia flavor works on GitHub actions.

Maybe another change that I'd propose is to make the Nvidia version the default one. I remember that in VRX we had to explicitly pass a `-n` argument for building and running the image.

M1chaelM, 2020-08-17:
Thanks @caguero . @bsb808 if there are no objections I'll take Carlos' suggestion.

BruceMty, 2020-08-17:
I have seen multiple configurations be a big burden.  If we can't avoid it, hopefully CI can help ease that burden.

M1chaelM, 2020-09-24:
Resolved by #38

closed 2020-09-24 by M1chaelM