Field-Robotics-Lab/dave #36: uuv_dave.launch fails if it does not find a controller

M1chaelM, 2020-08-23:
## Problem

If no controller is connected, or it is not at `/dev/input/js0` the following error appears at launch:
```
[ERROR] [1598203806.880535332]: Couldn't open joystick /dev/input/js0. Will retry every second.
```
This turns out to be fatal. Later we see:

```
[INFO] [1598203808.546280, 0.000000]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1598203808.547676, 0.000000]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1598203808.548790, 0.000000]: Loading controller: /rexrov/oberon7/finger_left_joint/controller
[ERROR] [1598203808.590425, 0.004000]: Spawn service failed. Exiting.
```

## Solution
Ideally we should check for a controller at each available `/dev/input/js#` and if none are found fall back to keyboard teleop control.

M1chaelM, 2021-02-01:
Closing because I'm no longer able to reproduce this error on Noetic or Melodic.

closed 2021-02-01 by M1chaelM

mabelzhang, 2021-02-02:
Related to looking for the correct number in `/dev/input/js#`:

This may or may not help, depending on whether everyone uses the same kind of joystick. Posting for future reference.

If the device ID is consistent (say, same brand, same model), you can use [udev](https://opensource.com/article/18/11/udev) rules to always get the right port based on the device ID.
The device ID looks something like `1d6b:0003` and is listed by the `lsusb` command, which shows a list of devices currently connected.

Once you have the device ID, you can create a one-liner `.rules` file to look for that ID and symlink to it from a known place, like `/dev/dave_controller`.

I have an example here https://github.com/mabelzhang/hri-safe-remote-control-system/pull/6
The one-liner in the example is
```
SUBSYSTEMS=="usb", SUBSYSTEM=="tty", KERNEL=="ttyACM[0-9]*", ACTION=="add", A=="2a99", =="d002", SYMLINK+="hri_vsc", MODE="666"
```
The first half of the device ID goes in the `ATTRS{idVendor}`, and the second half goes in the `ATTRS{idProduct}`.
The original port goes to `KERNEL`, and you might have to play around with the `SUBSYSTEMS` attributes.
The user does need to `sudo cp` the `.rules` file to `/etc/udev/rules.d/`, and reload the rules using `sudo udevadm control --reload-rules`.

Then the code can just look for `/dev/dave_controller` without specifying the actual port.

It can come in handy on the same computer, for example, when you have a device that shows up as `1` on the host, and `0` in a Docker container.