Field-Robotics-Lab/dave #45: inaccurate IMU sensor data

PenguineZhang, 2020-09-04:
I want to use IMU sensor data to estimate the velocity of the ROV. However, the IMU sensor on rexrov model doesn't produce accurate linear acceleration data, but not the case for angular velocity (all 0 when stationary). For example, the linear acceleration in X and Y direction fluctuates slightly at about -0.85 and 0.3 respectively, while linear acceleration in Z direction is about 9.75, which makes sense more than in X and Y direction,  when ROV remains still. What's the cause of such bias?

mabelzhang, 2020-09-04:
I am not particularly familiar with the IMU sensor. My first guess would be, could there be external forces acting on the robot when it's still, e.g. wind?

Since I'm gone next week, I'm tagging @caguero so that you can get a quicker answer.

PenguineZhang, 2020-09-04:
@mabelzhang @caguero Thanks for the tips! You are right, there's a current plugin enabled. However, the linear acceleration in X and Y direction is unchanged (-0.85 and 0.3) when simulation starts and changed to -1.08 and 0.36 when I moved the ROV and let it hover, given that I set the current velocity to be max of 0.00001.

mabelzhang, 2020-09-05:
Those numbers look very close. Assuming there isn't a bug (which Carlos would know more about), is it possible to look at what the current is? One guess is that the current could be increasing in some direction, and for the vehicle to stay in the same place, it has to increase effort in the opposite direction. Then it could show up as an increase in the acceleration but the actual velocity with respect to the world is not changing? All of this is based on Z being 9.75, which looks like it's countering gravity to stay in place.

Anyway, that's just my super naive guess, not having any experience with an IMU. I will let Carlos give a more informed answer.

M1chaelM, 2020-09-15:
Since we have opted not to use the IMU data, we can probably close this issue unless it really is a bug. @PenguineZhang do you agree?

PenguineZhang, 2020-09-15:
@M1chaelM I agree. Thanks.

closed 2020-09-15 by M1chaelM