Field-Robotics-Lab/dave #46: Specification of glider rigid body inertia values

bsb808, 2020-09-08:
Running this command
```
roslaunch uuv_dave uuv_gliders_hybrid_whoi.launch paused:=true
```
generates a warning...
```
[ WARN] [1599576113.958181542]: The root link glider_hybrid_whoi/base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
```

This suggests that the rigid body inertia of the vehicle may not be set.

woensug-choi, 2020-09-09:
This can be solved by adding following dummy joint to fix urdf format defined inertia.
```
    <joint name="dummy_joint" type="fixed">
      <parent link="dummy"/>
      <child link="base_link"/>
    </joint>
```

Besides, the inertial matrix definition is going to be recalculated by the glider kinematics plugin (which is being developed) with Fossen's theory at [line where inertial matrix is calculated in the plugin](https://github.com/Field-Robotics-Lab/glider_hybrid_whoi/blob/cf38957c2192e04ed535593708472742a9665614/hybrid_glider_gazebo_plugins/hybrid_glider_gazebo_plugins/src/HydrodynamicModel.cc#L438).

woensug-choi, 2020-09-25:
@M1chaelM @bsb808 
I would say this is an intermediate error issue and not to be included as 'should be resolved before 2.0 release'.

For issue being `This suggests that the rigid body inertia of the vehicle may not be set.`, my answer is `the inertial matrix definition is going to be recalculated by the glider dynamics plugin` which is being developed and will be merged with the glider vehicle models later.

Currently, the inertial definitions at urdf are required for UUV_Simulator's `uuv_underwater_object_ros_plugin`. Unless all the plugins and sensors are stripped off for glider models or new plugin should be used. I am working on a later solution with better dynamics for gliders targeting release 3.0.

woensug-choi, 2021-02-03:
Re-summarizing this subject.
Problem:
```
[ WARN] [1599576113.958181542]: The root link vehicle_name/base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
```
- Not fatal error but a warning message.
Reason:
UUV simulator's underwater object plugin (which rules hydrodynamic characteristics of the vehicles) has it's own way of defining vehicle hydrodynamics including inertia that is not generic methods of the gazebo. The plugin overwrites at the runtime.
You may find the same warning message with the UUV_simulator's launch commands,
`roslaunch uuv_gazebo rexrov_default.launch`
Also at the Dave's Rexrov,
`roslaunch uuv_dave uuv_dave.launch`

Conclusion:
It's complaining that the Gazebo may not understand vehicle inertial properties. But as long as the underwater object plugin of the UUV simulator is being used, the hydrodynamics properties including inertial properties are working in place. Therefore, the warning message can be and should be ignored for our cases.

closed 2021-02-03 by woensug-choi