Field-Robotics-Lab/dave #48: Reintegrate predator manipulator texture.

bsb808, 2020-09-08:
From PR#42

No texture: The individual links were not centralized. After I centralized them in Blender, the tag in the dae file responsible for texture disappears.

bsb808, 2021-02-02:
@M1chaelM - Would you be able to take a look at this one since it seems like you started the issue.  Not sure if there is anything to address.

M1chaelM, 2021-02-02:
@bsb808 Sure I can look at it but I did not start it. It's from Youssef. I think the problem was that he did not get around to adding the texture to the predator arm, though I'm not sure why.

M1chaelM, 2021-02-18:
Not sure what has changed, but it seems that the texture is showing up, but is not aligned correctly. Running:

```
roslaunch uuv_dave uuv_predator.launch
```
Spawns a uuv with a predator arm attached, but some sections floating:

[image]

The wrist joint hovers above the main arm:
[image]

The end effector is back behind it:

[image]

And the fingers in the end effector are misaligned and coming out the wrong side:

[image]

M1chaelM, 2021-02-19:
In addition to the problems above, it appears that none of the texture png files are actually checked in to the repository.

M1chaelM, 2021-02-19:
To reconstruct what this model is supposed to look like (with texture), I took the following steps:
* Re-download all the original `.dae` files provided by designer.
* Download the `Predator.png` texture and rename to `predator.png`.
* Edit all the original `.dae` files to reference `predator.png` in the `init_from` tag. E.g.:
```
<init_from>predator.png</init_from>
```
* Check all `.dae` files in meshlab.

As an example, the file `Predator-forearm.dae` looks like this:

[image]

M1chaelM, 2021-02-19:
## Notes
* All the original `.dae` files began with the prefix `Predator`. These were renamed when the model was added.
* The mapping between the original files and the renamed files is as follows:
|Original|Renamed|
|-----------|------------|
|Predator-root.dae |base.dae|

* claw_left.dae
* claw_right.dae
* forearm.dae
* hand.dae
* knuckle.dae
* shoulder.dae
* tip_left.dae
* tip_right.dae
* upperarm.dae
Predator-base.dae
Predator-claw-left.dae
Predator-claw-right.dae
Predator-forearm.dae
Predator-hand.dae
Predator-knuckle.dae
Predator-smallarm.dae
Predator-tip-left.dae
Predator-tip-right.dae
Predator-upperarm.dae

* `Predator-base-bigarm.dae` and `Predator-smallarm.dae` load without error in meshlab but display nothing.
* `Predator.dae` is the entire model.

M1chaelM, 2021-02-23:
The above alignment can be resolved by editing the original `.dae` files but replacing the `<matrix sid="transform">` tag values with the values from the corresponding `.dae` file that was "centralized" (and otherwise modified) by blender. Doing so for all tags and rerunning the demo **in Gazebo 9** produces the correct predator arm, with the exception that the upper arm is much too big:

[image]

I'm not sure what causes this, but it can be fixed by adding the attribute `scale="0.7 0.7 0.4"` to the mesh tag in the `serial_arm_predator.xacro` file:

[image]

M1chaelM, 2021-02-23:
Controls now function properly. Not sure if these are documented anywhere, so I will add them to the tutorial. The right back button on the LogiTech controller (RB) has to be depressed to enable manipulator control mode. 

Right Joystick
|Key |Action|
|-----------|------------|
| Down | Shoulder joint lowers |
| Up | Shoulder joint raises |
|Left | Shoulder joint swivel left |
|Right| Shoulder joint swivel right |

Left Joystick
|Key |Action|
|-----------|------------|
| Down | Elbow joint lowers |
| Up | Elbow joint raises |
|Left | Wrist joint swivel left |
|Right| Wrist joint swivel right |

4-way keypad
|Key |Action|
|-----------|------------|
| Down | Wrist joint lowers |
| Up | Wrist joint raises |

XYAB Buttons
|Key |Action|
|-----------|------------|
| B | Claw opens |
|X | Claw closes |

mabelzhang, 2021-02-23:
Hmm rescaling to 0.7 seems a hack. Do you think this is a problem with the code in this repo or something to do with Gazebo 9 vs. 11?

I thought maybe it might be a difference in the version of sdformat specification - there are some recent fixes to [sdformat](https://github.com/osrf/sdformat) (e.g. xyzOffset, rpyOffset), and we've been adding [new features](https://www.youtube.com/watch?v=V6ZOumG8lFU) to 1.9, but I'm not seeing those tags or versions used in this repo. So probably irrelevant.

I did notice that all the world files in `uuv_dave/worlds` are using version 1.4, while all the models in `uuv_dave/models` are using 1.6 (though these aren't used in this launch file), and the `predator_description` xacro files don't specify a version. In that case, I guess it's loading 1.4 for the arm. Just an observation. Might not mean much.

Another version difference is the sdformat library version, which is sdformat9 on my Ubuntu 20.04 and sdformat6 on my 18.04.

I don't know. Maybe it's irrelevant. If you become convinced at some point that it's caused by something that changed in Gazebo 11 or sdformat9, it'd be good to know so that we can fix it on our end.

M1chaelM, 2021-02-24:
@mabelzhang Thanks for looking this over! The rescaling was weirdly necessary in both Gazebo 9 and 11. I'm going to try to figure out what's going on.

Meanwhile, the fix here only fixes things in Gazebo 9. In Gazebo 11 the wrist and hand are detached again. I'm going to create a separate issue for this because it definitely looks like it occurs because of the change in versions. I think I may have some relevant error messages I can add when I put it up.

Also, the inertia is not looking too good in either version
[image]

I will make an issue for that also.
:

M1chaelM, 2021-02-24:
@mabelzhang OK I will able to eliminate the need for the scaling tag by editing the .dae file. On reflection, I don't think the scaling tag was really a hack. I think the original upperarm piece we designed just had the wrong scale and Youssef fixed it by rescaling in blender, which causes the correct scale to get baked into the `.dae` file. Unfortunately when he did this he deleted the texture tags. Anyway, it's better for the `.dae` file to have the right scale, but I think the cause of this problem is independent of both Gazebo and our code for assembling the arm.

M1chaelM, 2021-02-24:
See issue #97 for problems that are unique to Gazebo 11.

M1chaelM, 2021-07-29:
This was resolved with PR #99 and should be closed.

closed 2021-07-29 by M1chaelM