Field-Robotics-Lab/dave #59: cpplint.py header guard suggests are inconsistent

M1chaelM, 2020-09-16:
## Current behavior:

When run locally, cpplint.py recommends using the full path of the file from the root of the repository to name the guard. So, for example, `code_check.sh` will include an error like this:
```
./include/nps_uw_sensors_gazebo/nps_gazebo_ros_gpu_sonar_single_beam.h:19:  
#ifndef header guard has wrong style, please use: 
INCLUDE_NPS_UW_SENSORS_GAZEBO_NPS_GAZEBO_ROS_GPU_SONAR_SINGLE_BEAM_H_
```
This error can be eliminated by changing the header guard as requested (although this is not ideal). However, when the same check is performed in the github action environment, the file path and corresponding variable name is very strange:
```
./include/nps_uw_sensors_gazebo/nps_gazebo_ros_gpu_sonar_single_beam.h:19:
 #ifndef header guard has wrong style, please use: 
___W_NPS_UW_SENSORS_GAZEBO_NPS_UW_SENSORS_GAZEBO_INCLUDE_NPS_UW_SENSORS_GAZEBO_NPS_GAZEBO_ROS_GPU_SONAR_SINGLE_BEAM_H_
```

Both of these are wrong, and it is not possible pass tests locally and in the github action environment with the same code.

## Additional Notes
* Although the `code_check.sh` script specifies file extension `.hh` should be checked, `cpplint.py` only runs the `CheckForHeaderGuard` function on files with a `.h` extension ([the relevant code is here](https://github.com/Field-Robotics-Lab/dave/blob/008a38b6df6a9241515302ed7439eedb318ec240/tools/cpplint.py#L5409)).
* This means this issue does not show up for files with `.hh` extensions because it just skips the header guard check entirely.
* I believe this means that this check was never actually run on any code in vrx because we didn't write any headers with the .h extension, so this problem never showed up.
* Conversely, a workaround is just to use the `.hh` extension and skip this check.

caguero, 2020-09-25:
I made some changes in pull request #72 and the style checker passes locally and on GitHub.Typically, I only run:

```
sh tools/code_check.sh
```

Is this still an issue?

bsb808, 2021-07-27:
@M1chaelM Is this still an issue?

M1chaelM, 2021-07-28:
> I made some changes in pull request #72 and the style checker passes locally and on GitHub.Typically, I only run:
> 
> ```
> sh tools/code_check.sh
> ```
> 
> Is this still an issue?

Yes, it turns out this is still a problem. For some reason all the header guard suggestions start with `___W`.

I was confused about why these problems seemed to go away for a while and have recently come back during the reorganization, but now I think I've figured this out. I believe our checker only looks at guards for headers that end in ".h". I think we made these errors disappear in the past by changing our header file extensions to use either ".hh" or ".hpp". This just causes the check not to run because the checker doesn't expect these extensions. So, obviously, not the best way of passing the check.

As a separate issue that I think came up before: for our collective sanity we probably shouldn't have headers ending with all three of these file extensions.

M1chaelM, 2021-07-29:
Adding a comment @mabelzhang made in #129:

> Nice work guys! I'm not quite awake yet, but just wanted to note for future reference re header guards if you ever want to  enable it again. Maybe changing the configuration to just do the file name itself, not the full path, is a better option. That's how we name our header guards in ROS and Gazebo.

Mabel, do you know how to do that? I'm not a  `cpplint.py` power user but I don't see an option for this in our script.

woensug-choi, 2021-07-30:
Addressed at https://github.com/Field-Robotics-Lab/dave/pull/139

bsb808, 2021-07-30:
@M1chaelM and @caguero 
We have this labeled as to be addressed for 4.0.0.   How should we proceed?

M1chaelM, 2021-07-30:
Per @woensug-choi 's comment above, this is resolved with PR #140 and can be closed when it is merged.

bsb808, 2021-07-30:
Closing b/c PR#140 is merged.  Nice work guys.

closed 2021-07-30 by bsb808