Field-Robotics-Lab/dave #61: PR#50 Post-merge to do

bsb808, 2020-09-17:
# USBL Dev - first PR

https://github.com/Field-Robotics-Lab/dave/pull/50

B/c the PR is in github and Peng can't make changes to github I am doing the following...

1. Where possible, will try to make changes directly to the code in the PR.  Then Peng will review the PR (test it) and approve.  Once approved I will merge on github.

2. Below are my comments on the PR that should be addressed post-merge.  They will be addressed on the gitlab side so that the tools can support interactive review.

## #include <opencv2/imgproc/imgproc_c.h>

The addition in
 `sonar_dave_plugins/sonar_dave_uuv/src/gazebo_ros_image_sonar.cpp `

This is an unrelated change.  I have removed it from the PR.

If it is still an issue with github/master, it should be a separate PR.  

## File extensions

Believe we decided on .cc and .hh 

## Move package to nps_us_sensors_gazebo

## Message definition comments

Please add documentation in the .msg files to clarify the interface.  For example see the ROS sensor_msgs, e.g., https://github.com/ros/common_msgs/blob/noetic-devel/sensor_msgs/msg/NavSatFix.msg

## Verification

If you have verified all this, no need to reply, but wanted to make sure we have verified the various functionalities, e.g, 

* Have we verified the two different interrogation modes.
* Have you verified that the sphericial and cartesian outputs behave as expected?  I looked at the math briefly, but would need to run tests in various quadrants and orientations to make sure all the details are correct.

## Ping Scheduler

This appears to be a constant update rate - true?

If so, would recommend a different idiom for used.  Look at the way other gazebo plugins implement an update rate within the ModelPlugin::Update() method.  The use of a separate timer object and thread seems to overly complicate the implementation.

## Set sound speed directly

I believe it would be simpler to set the speed of sound directly - rather than indirectly through the temperature.

Also, speed of sound should probably be an SDF parameter of the plugins rather than set via ROS messaging.

## Documentation

For users we should explain a bit more about how this plugin works - what is the theory of operation?  One thing that would help would be to sketch the communications - maybe use the rosgraph as a start and explain the ROS communication.

PenguineZhang, 2020-09-25:
@BruceMty @bsb808 The changes are made and pushed to Gitlab. PR is submitted.  When you get a chance please take a look

bsb808, 2021-02-01:
Addressed with PR#50 https://github.com/Field-Robotics-Lab/dave/pull/50

closed 2021-02-01 by bsb808