Field-Robotics-Lab/dave #73: conflicting interpretations of gpu_ray sensor parameters

M1chaelM, 2020-09-25:
## Context
We use the gpu_ray sensor to approximate our under water lidar sensor, which is based on the specs for the 3d at Depth SL3. By default, the sensor field of view is 30 degrees x 30 degrees and contains 1450 x 1450 points. We produce this by setting the following values for both the `<horizontal>` and `<vertical>` elements of the `<scan>` element:

```
<samples>1450</samples>
<resolution>0.1</resolution>
<min_angle>-${pi/12}</min_angle>
<max_angle>${pi/12}</max_angle>
```

## Problem

On some systems, this produces the desired effect:
[image]

On other systems, however, this same configuration produces very strange laser effects:
[image]

## Notes
We don't currently know what differences between systems could be causing this effect. Of particular note is that in at least one case the plugin works correctly inside a Docker install, but fails on a host-based install on the same machine.

M1chaelM, 2021-02-02:
I am currently unable to reproduce this issue on the laptop where it first appeared. The laptop has been upgraded from Ubuntu 18.04 to 20.04. Running 18.04 in a Docker container also produces the expected behavior (as it did previously). We have also tested on another laptop with identical hardware running 18.04 installed on the host, but no longer see this behavior.

@dtdavi1 I think you had the only other machine where this issue occurred. Could you check to see whether it's still a problem? I think this was a driver issue and was resolved with a driver update, so if you don't see it anymore I'll close the issue. To test, build the latest and run:

```
roslaunch nps_uw_sensors_gazebo uw_lidar_standalone.launch
```
Once it launches, hit play to make sure you see the normal-looking beam swing to face the cylinder. If it freezes or if you get a black sphere that means the bug is still bugging us.

dtdavi1, 2021-02-02:
@M1chaelM Will do.  I probably can't get to it today because of class and meetings but should be able to confirm it tomorrow (no reason to believe that it's fixed itself).

dtdavi1, 2021-02-03:
@M1chaelM It's working fine for me now, so you can probably close the issue (at least until it pops up on someone else's system).  I think that I might have had an out of date library (I got a warning in another context about Ignition being out of date, so that's my best guess).  I actually couldn't get it to run at all when I first tried, so I uninstalled pretty much everything (Gazebo, ROS, etc.) and then built everything back up from the ground.  After doing that, it works correctly.

M1chaelM, 2021-02-03:
Thanks @dtdavi1 ! OK I think we've done due diligence for now so I'm closing this.

closed 2021-02-03 by M1chaelM