Field-Robotics-Lab/dave #77: GPS latitude and longitude seem to have their signs flipped

xydesa, 2020-10-13:
* `roslaunch uuv_dave uuv_dave.launch`
* `rostopic echo /rexrov/gps`
```bash
header: 
  seq: 641
  stamp: 
    secs: 21
    nsecs: 830000000
  frame_id: "rexrov/gps_link"
status: 
  status: 0
  service: 1
latitude: -56.7189407714
longitude: -3.51555967066
altitude: -93.0446776133
position_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
position_covariance_type: 3
---
```
* The signs on both latitude and longitude are flipped from what is defined in the world file: https://github.com/Field-Robotics-Lab/dave/blob/master/uuv_dave/worlds/uuv_dave_ocean_waves.world#L38
* They also seem wrong given the comment in the world file
  * [-56.7189407714, -3.51555967066](https://www.google.com/maps/place/56%C2%B043'08.2%22S+3%C2%B030'56.0%22W/@-40.6017017,0.1756212,3z/data=!4m5!3m4!1s0x0:0x0!8m2!3d-56.7189408!4d-3.5155597)
  * [56.7189407714, 3.51555967066](https://www.google.com/maps/place/56%C2%B043'08.2%22N+3%C2%B030'56.0%22E/@55.1902927,3.9973646,5.65z/data=!4m5!3m4!1s0x0:0x0!8m2!3d56.7189408!4d3.5155597)

This seems to be caused by this PR in the uuv_simulator repo: https://github.com/uuvsimulator/uuv_simulator/pull/250
In that PR, the maintainer stated that the long term fix was to replace the sensor plugin completely with one that does the necessary computation independent of Gazebo.

bsb808, 2020-10-14:
Thank you for the report.  Given our applications, we haven't paid much attention to the GPS publications.  It sounds like switching to the standard hector GPS plugin would be a better option for use-cases that would have satellite access.

We'll leave it on the list, but it will be a low priority.  Happy to review a PR from the community.

woensug-choi, 2020-12-21:
I think the opposite signs are correct as they are supposed to be. Since the origin moved from `0.0`, `0.0` to `56.71897669633431` and `3.515625`. The location of the vehicle is now at `-56.71897669633431`, `-3.515625`.

The issue on my hand is that the value is slightly off from what it's supposed to be.
- Expected value
  `-56.71897669633431`, `-3.515625`
- GPS sensor reporting value
  `-56.7189407714`, `-3.51555967066`

bsb808, 2021-02-02:
Addressed in https://github.com/Field-Robotics-Lab/dave/pull/94  - close issue when/if PR is merged.

bsb808, 2021-02-08:
Addressed in #94

closed 2021-02-08 by bsb808