Field-Robotics-Lab/dave #85: Fix hard-coded path in stratified current demo launch file

dtdavi1, 2021-01-06:
In order to run the demo, the path to the .csv file specified in the launch file (uuv_dave/uuv_dave_transientcurrent_demo.launch) must be manually edited to a valid path on the local machine.  This needs to be changed to a relative path that the ROS or Gazebo "find" functionality can locate.

woensug-choi, 2021-01-29:
Jessica might work on this subject as a practice :)

j-herman, 2021-02-01:
I have this working locally.  If anyone needs it soon-ish, I can create a pull request, but otherwise I'll send it up when I have a working two-vehicle demo.

j-herman, 2021-02-03:
Ok, per recommendations/ comments I have updated the code.  I'm going to cancel the existing pull request since I'm now working in a branch on this repo.
The code will allow for three options:
1.  Specify the filename in the <database> tag.  The file must be located in the uuv_dave/worlds folder
2.  Specify the full path to the file in the <database> tag
3.  Comment out the database tag altogether, and the default /uuv_dave/worlds/transientOceanCurrentDatabase.csv will be loaded

j-herman, 2021-02-08:
Recommend closing based on approved PR #95

j-herman, 2021-02-08:
Closing per 2/8 discussion; feel free to reopen if needed

closed 2021-02-08 by j-herman