Field-Robotics-Lab/dave #86: Remove plugin cross-dependency in transient ocean current plugin.

dtdavi1, 2021-01-06:
The single ocean current value published by the transient current plugin (uuv_dave/UnderwaterCurrentROSPlugin.cc) is dependent on a separate model plugin to select the depth for which the depth is published.  Since the plugin has been updated to publish an array of currents for all of the depths, the single value no longer needs to be depth-specific.  Reversing this change will maintain reverse compatibility with models that assume a single homogeneous ocean current value.

woensug-choi, 2021-02-03:
@dtdavi1 I remember you saying that you are working on this. If you are not, or if you need help, please let me know!

dtdavi1, 2021-02-03:
@woensug-choi Yes, I've got this one for action.  It's actually done for the most part (it's not "multi-vehicle" ready, though), but I haven't pushed the commit yet.

closed 2021-02-15 by woensug-choi