Field-Robotics-Lab/dave #103: Can't build Dockerfile; Invalid Signature from Cuda repo

M1chaelM, 2021-03-08:
I recently had to clear out cached Docker images to make space for the massive Mathworks build process (which didn't work). Now, attempting to rebuild our Dave Docker image with:
```
./build.bash .
```
I get to step 5 in the build process and then it crashes:
```
W: GPG error: https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64  Release: The following signatures were invalid: BADSIG F60F4B3D7FA2AF80 cudatools <cudatools@nvidia.com>
E: The repository 'https://developer.download.nvidia.com/compute/cuda/repos/ubuntu1804/x86_64  Release' is not signed.
```
This is a known issue: https://github.com/NVIDIA/nvidia-docker/issues/1369 but it is closed because `nvidia-docker` doesn't want to own it. They want it submitted to their developer forum. It has been: https://forums.developer.nvidia.com/t/cuda-10-1-install-gpg-error-the-following-signatures-were-invalid-badsig-f60f4b3d7fa2af80/79944

It looks like there are workarounds, but maybe the best fix is to move to using Rocker from a standard 18.04 base image.

woensug-choi, 2021-03-19:
Is this the link to the Rocker you meant? (https://github.com/osrf/rocker)

M1chaelM, 2021-07-29:
We have moved to using rocker so I am closing this.

closed 2021-07-29 by M1chaelM