Field-Robotics-Lab/dave #107: Bathymetry tiles location

woensug-choi, 2021-03-24:
Bathymetry tiles are products of converting large tiff data into tiles using [Bathymetry_Converter](https://github.com/Field-Robotics-Lab/Bathymetry_Converter). The converter is kept in separate repo since it takes a long time and not easy to install ([tutorial](https://github.com/Field-Robotics-Lab/dave/wiki/Bathymetry-Data-Conversion-for-Bathymetry-Plugin)). In the tutorial, the original tiff (ncei19_n41x75_w070x75_2018v1.tif) is downloaded from my personal dropbox link using wget function. (`wget -O bathymetry_source.tar.bz2 https://www.dropbox.com/s/x3acdnnpw3ej9b4/bathymetry_source.tar.bz2?dl=1`).

Currently, the minimal tiles to run the plugin correctly are included in the dave repo. However, we don't want any static example data to be included in the repo. The products of such should not be stored within the git repos.  Instead, it should be kept in some persistent storage and providing instructions on how (e.g., use a curl/wget script to download from gdrive/Box/dropbox).

So what we propose is,
1. **Remove everything out of the repo (Proposed)**
   - **Remove Launch file (bathy_dave.launch), world file (dave_bathy.world), and xacro file (dave\uuv_dave\urdf\rexrov_oberon7_with_bathy.xacro), and all tile models from dave repo**
   - **Make those removed files downloadable (with minimal tiles to launch tutorial case) by wget (hosted on Google Drive)**
   - Rest of the tiles are not provided. They are DIY following the Bathymetry Converter and its tutorial
2. Leave minimal examples in the repo (Current setting)
   - Leave Launch file (bathy_dave.launch), world file (dave_bathy.world), xacro file (dave\uuv_dave\urdf\rexrov_oberon7_with_bathy.xacro), and minimal tiles in dave repo
   - Rest of the tiles are not provided. They are DIY following the Bathymetry Converter and its tutorial

@bsb808 Is the proposed method what you meant?

bsb808, 2021-07-27:
@woensug-choi and @dtdavi1 is this sorted out in the v3.1_reorg?  Should we address this for 4.0.0 or push it right in the schedule?

woensug-choi, 2021-07-28:
@bsb808 I will make commit and push for above `proposed` method. In the same sense, the `hybrid_glider_whoi` consists of massive bathymetry tiles of the Buzzards Bay. I suggest removing the `hybrid_glider_whoi` repo from vcstool dependencies. By excluding the glider repo, it will act as a downloadable dataset that is not included in the mainstream `dave` repo.

bsb808, 2021-07-28:
@woensug-choi Sounds great - thank you.

woensug-choi, 2021-07-30:
Made a single command to download files necessary to run bathymetry demo
```
wget -qO- https://bit.ly/3j1rJik | tar xvf -
```
The link is shortened using bitly and the files are hosted at [gdrive/Dave/DaveExamples/bathymetry_demo.tar](wget -qO- https://bit.ly/3j1rJik | tar xvf -)

[Bathymetry Wiki](https://github.com/Field-Robotics-Lab/dave/wiki/Bathymetry-Integration) is modified accordingly.

closed 2021-07-30 by woensug-choi