Field-Robotics-Lab/dave #108: HDT arm

wangcongrobot, 2021-04-15:
Hi, thanks for the really great project! 

From the [roadmap](https://github.com/Field-Robotics-Lab/dave/wiki/roadmap), 

> Release 3.0: Addition of new model based on HDT Adroit.

Is it available now?

bsb808, 2021-04-17:
Not yet - are you using that arm?  It would help to have a collaborator with access to the hardware.

wangcongrobot, 2021-04-17:
Yes. We have a hdt underwater arm, but not test a lot.

wangcongrobot, 2021-04-17:
I'm happy to collaborate and we have several kinds of underwater arms that can be used. Hope to contribute to the great project Dave. We also use uuvsimulator for our simulations.

crvogt, 2021-04-20:
@wangcongrobot Are you part of the CDT at UoE or Heriot-Watt?

wangcongrobot, 2021-04-20:
@crvogt Yes. In the ocean system lab of Heriot-Watt and we have an HDT underwater arm for testing. But it will be sent back later. The [blueprintlab of reach5mini and bravo](https://blueprintlab.com/) we bought are also easy to use. But the ROS driver is not perfect now.

crvogt, 2021-12-09:
Closing this. @wangcongrobot Be sure to occasionally check in, we're still expanding our manipulator work.

closed 2021-12-09 by crvogt