Field-Robotics-Lab/dave #109: Error during building

tfabbri, 2021-05-18:
Dear All, 

I am facing some issues during the compile of this project. 
I am considering the branch `tidal_current_fix_before_reorg` based on the indications of the repository [Glider Hybrid Whoi](https://github.com/Field-Robotics-Lab/glider_hybrid_whoi). 
During the execution of the `catkin_make` command, I have the following error:

```
[ 97%] Building CXX object dave/usbl_gazebo/CMakeFiles/transceiverPlugin.dir/src/transceiverPlugin.cpp.o
In file included from /home/tommaso.fabbri/Git/aasw/passiveWorkspace/gitAsw_pSimulation/uuv_ws/src/dave/usbl_gazebo/src/transceiverPlugin.cpp:19:0:
/home/tommaso.fabbri/Git/aasw/passiveWorkspace/gitAsw_pSimulation/uuv_ws/src/dave/usbl_gazebo/include/transceiverPlugin.hpp:16:10: fatal error: ignition/common/StringUtils.hh: No such file or directory
 #include <ignition/common/StringUtils.hh>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
dave/usbl_gazebo/CMakeFiles/transceiverPlugin.dir/build.make:62: recipe for target 'dave/usbl_gazebo/CMakeFiles/transceiverPlugin.dir/src/transceiverPlugin.cpp.o' failed
make[2]: *** [dave/usbl_gazebo/CMakeFiles/transceiverPlugin.dir/src/transceiverPlugin.cpp.o] Error 1
CMakeFiles/Makefile2:11022: recipe for target 'dave/usbl_gazebo/CMakeFiles/transceiverPlugin.dir/all' failed
make[1]: *** [dave/usbl_gazebo/CMakeFiles/transceiverPlugin.dir/all] Error 2
```
The `libignition-common-dev` is installed on my system through apt. 
My system is Ubuntu 18.04 with ROS Melodic. Any suggestions?

Thank you

woensug-choi, 2021-05-19:
Never seen those compile errors. Did you do `sudo apt upgrade libignition-math2` ?

tfabbri, 2021-05-19:
@woensug-choi thank you for your answer. 
Unfortunately it is at the latest version:
```
libignition-math2 is already the newest version (2.9.0+dfsg1-1)
```
Any suggestion to debug this?

closed 2021-07-01 by woensug-choi