Field-Robotics-Lab/dave #116: Catkin_make fails, libignition-math2 package not found

woensug-choi, 2021-07-22:
On WSL Ubuntu 20.04, `sudo apt-get install -y libignition-math2` fails because `libignition-math2` is only at package manager up to Ubuntu 18.04. Somewhere following along with the installation wiki, `libignition-math6` is installed.

`catkin_make` fails on Ubuntu 20.04 with ROS Noetic Gazebo 11 and I suspect the above problem is causing it. Are we suppose to git clone some other specific branch or version of `uuv_simulator`?

some of `catkin_make` fail error logs,
```
In file included from /usr/include/ignition/math6/ignition/math/Spline.hh:22,
                 from /usr/include/gazebo-11/gazebo/common/Animation.hh:22,
                 from /usr/include/gazebo-11/gazebo/common/common.hh:5,
                 from /usr/include/gazebo-11/gazebo/gazebo_core.hh:19,
                 from /usr/include/gazebo-11/gazebo/gazebo.hh:20,
                 from /home/woensug/uuv_ws/src/uuv_simulator/uuv_gazebo_plugins/uuv_gazebo_plugins/src/Dynamics.cc:16:
/usr/include/ignition/math6/ignition/math/Helpers.hh:40:13: warning: variable templates only available with ‘-std=c++1
’ or ‘-std=gnu++14’
   40 | constexpr T IGN_MASSMATRIX3_DEFAULT_TOLERANCE = T(10);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[  5%] Generating C++ code from ds_acomms_msgs/Point_F32.msg
/usr/include/ignition/math6/ignition/math/Helpers.hh: In function ‘std::chrono::_V2::steady_clock::duration ignition::math::v6::stringToDuration(const string&)’:
/usr/include/ignition/math6/ignition/math/Helpers.hh:990:28: error: ‘chrono_literals’ is not a namespace-name
  990 |       using namespace std::chrono_literals;
      |                            ^~~~~~~~~~~~~~~
/usr/include/ignition/math6/ignition/math/Helpers.hh: In function ‘std::chrono::_V2::steady_clock::time_point ignition::math::v6::stringToTimePoint(const string&)’:
/usr/include/ignition/math6/ignition/math/Helpers.hh:1032:28: error: ‘chrono_literals’ is not a namespace-name
 1032 |       using namespace std::chrono_literals;
      |                            ^~~~~~~~~~~~~~~
/usr/include/ignition/math6/ignition/math/Helpers.hh:1033:56: error: unable to find numeric literal operator ‘operator""s’
 1033 |       std::chrono::steady_clock::time_point timePoint{-1s};
      |                                                        ^~
/usr/include/ignition/math6/ignition/math/Helpers.hh:1033:56: note: use ‘-fext-numeric-literals’ to enable more built-in suffixes
/usr/include/ignition/math6/ignition/math/Helpers.hh:1033:58: error: no matching function for call to ‘std::chrono::time_point<std::chrono::_V2::steady_clock, std::chrono::duration<long int, std::ratio<1, 1000000000> > >::time_point(<brace-enclosed initializer list>)’
 1033 |       std::chrono::steady_clock::time_point timePoint{-1s};
```
```
In file included from /usr/include/ignition/math6/ignition/math.hh:30,
                 from /usr/include/sdformat-9.5/sdf/Param.hh:33,
                 from /usr/include/sdformat-9.5/sdf/Element.hh:28,
                 from /usr/include/sdformat-9.5/sdf/Actor.hh:25,
                 from /usr/include/sdformat-9.5/sdf/sdf.hh:2,
                 from /usr/include/gazebo-11/gazebo/common/Battery.hh:25,
                 from /usr/include/gazebo-11/gazebo/common/common.hh:8,
                 from /usr/include/gazebo-11/gazebo/gazebo_core.hh:19,
                 from /usr/include/gazebo-11/gazebo/gazebo.hh:20,
                 from /home/woensug/uuv_ws/src/uuv_simulator/uuv_gazebo_plugins/uuv_gazebo_plugins/src/Dynamics.cc:16:
/usr/include/ignition/math6/ignition/math/Capsule.hh: At global scope:
/usr/include/ignition/math6/ignition/math/Capsule.hh:91:20: error: ‘optional’ in namespace ‘std’ does not name a template type
   91 |       public: std::optional< MassMatrix3<Precision> > MassMatrix() const;
      |                    ^~~~~~~~
/usr/include/ignition/math6/ignition/math/Capsule.hh:91:15: note: ‘std::optional’ is only available from C++17 onwards
   91 |       public: std::optional< MassMatrix3<Precision> > MassMatrix() const;
```
```
In file included from /usr/include/ignition/math6/ignition/math/Spline.hh:22,
                 from /usr/include/gazebo-11/gazebo/common/Animation.hh:22,
                 from /usr/include/gazebo-11/gazebo/common/common.hh:5,
                 from /usr/include/gazebo-11/gazebo/gazebo_core.hh:19,
                 from /usr/include/gazebo-11/gazebo/gazebo.hh:20,
                 from /home/woensug/uuv_ws/src/uuv_simulator/uuv_gazebo_plugins/uuv_gazebo_plugins/src/Dynamics.cc:16:
/usr/include/ignition/math6/ignition/math/Helpers.hh: In instantiation of ‘std::tuple<_Tps ...> ignition::math::v6::breakDownDurations(DurationIn) [with Durations = {std::chrono::duration<long unsigned int, std::ratio<86400, 1> >, std::chrono::duration<long int, std::ratio<3600, 1> >, std::chrono::duration<long int, std::ratio<60, 1> >, std::chrono::duration<long int, std::ratio<1, 1> >, std::chrono::duration<long int, std::ratio<1, 1000> >}; DurationIn = std::chrono::duration<long int, std::ratio<1, 1000000000> >]’:
/usr/include/ignition/math6/ignition/math/Helpers.hh:829:57:   required from here
/usr/include/ignition/math6/ignition/math/Helpers.hh:810:29: error: no matching function for call to ‘get<std::chrono::duration<long unsigned int, std::ratio<86400, 1> > >(std::tuple<std::chrono::duration<long unsigned int, std::ratio<86400, 1> >, std::chrono::duration<long int, std::ratio<3600, 1> >, std::chrono::duration<long int, std::ratio<60, 1> >, std::chrono::duration<long int, std::ratio<1, 1> >, std::chrono::duration<long int, std::ratio<1, 1000> > >&)’
  810 |         (std::get<Durations>(retval) =
      |          ~~~~~~~~~~~~~~~~~~~^~~~~~~~
```

woensug-choi, 2021-07-22:
@crvogt Not being able to install libignition-math2 with apt but https://github.com/Field-Robotics-Lab/dave/wiki/Install-Directly-on-Host-(Noetic) has a line to do so. I think I am missing something here.

bsb808, 2021-07-27:
Is this specific to WSL?  We aren't yet 'officially' supporting WSL, so I'm not including it in our 4.0.0 milestone.  Let me know if that is incorrect.

woensug-choi, 2021-07-28:
It was a problem of using wrong dependencies. Changing `uuv_simulator` and `uuv_manipulator` at Field-Robotics-Lab with `noetic_updates` branch solved this issue.