Field-Robotics-Lab/dave #117: usbl_gazebo tutorial.world crashes when renaming box model of usbl_transceiver

tiko5000, 2021-07-22:
I experience unexpected behavior in the usbl tutorial when I rename the `<model name="box">`  in `tutorial.world` of `usbl_gazebo`

Steps to reproduce:

1. In `tutorial.world` rename `<model name="box">` to `<model name="box_2">`
2. `roslaunch usbl_gazebo tutorial.launch`
3. `rostopic pub --once /USBL/common_interrogation_ping std_msgs/String "data: 'ping'"`

Error message in console of `roslaunch usbl_gazebo tutorial.launch` 
```
[ INFO] [1626964248.162015610]: Physics dynamic reconfigure ready.
gzserver: /usr/include/boost/smart_ptr/shared_ptr.hpp:734: typename boost::detail::sp_member_access<T>::type boost::shared_ptr<T>::operator->() const [with T = gazebo::physics::Model; typename boost::detail::sp_member_access<T>::type = gazebo::physics::Model*]: Assertion `px != 0' failed.
Aborted (core dumped)
```

Can anybody reproduce this behavior?

mabelzhang, 2021-11-06:
This is expected behavior. The name `box` is hardcoded in the transponder plugin here
https://github.com/Field-Robotics-Lab/dave/blob/b50e2360fccee8f2447a36fe36a97bfd41f37b39/gazebo/dave_gazebo_model_plugins/src/usbl_transponder_plugin.cc#L223

and here

https://github.com/Field-Robotics-Lab/dave/blob/b50e2360fccee8f2447a36fe36a97bfd41f37b39/gazebo/dave_gazebo_model_plugins/src/usbl_transponder_plugin.cc#L243

Since it's a tutorial, I would suggest if you change the content of the world file locally, then you should change the code locally accordingly.

bsb808, 2021-12-15:
Closing with explanation by @mabelzhang

closed 2021-12-15 by bsb808