Field-Robotics-Lab/dave #119: Installation tutorials update for Noetic release

mabelzhang, 2021-07-24:
Looking for feedback for a few things I noticed about the installation tutorial while testing on Noetic.

I can update the tutorials once I get a green light. Just wanted to double-check these things aren’t already somewhere that I didn’t know about, since I’m still kind of new to the repo.

- [ ] Rocker’s most recent release, 0.2.3, is from Nov 2020. If we install via `apt` like the DAVE Noetic wiki says, the newer features like `--volume` aren’t available. I had problems using it because it only mounts `/home`, but none of my code is in my home directory (all my stuff is in a separate partition which I usually mount with `docker run --volume`). We might want to either ask Tully to make a new release, or put a note in the DAVE wiki saying if they want new features, then they need to install Rocker following the `venv` instructions on the Rocker page (that gets the latest version, and it worked for me).

- [ ] The [Noetic docker tutorial](https://github.com/Field-Robotics-Lab/dave/wiki/noetic_docker) doesn’t have instructions on how to build, while the Melodic page and the host installation page do. Since they’re all linked from the [installation page](https://github.com/Field-Robotics-Lab/dave/wiki/Installation), I propose to move the “Install Dave and dependencies from source” section from the Melodic page to the Installation page. All 4 ways of installation need to use these compiler commands.

- [ ] Is the top-level README intentionally vague? The Wiki has a wealth of information, but it’s not discoverable. I propose putting a sentence in the top-level README that links to the wiki.

- [ ] Dependencies `ds_sim`, `ds_msgs`, `uuv_simulator`, `uuv_manipulators`, especially our custom forks for uuv_* (currently in `noetic_updates` branches but presumably will be in `master` at the time of release?), should be documented, e.g. in a new “Dependencies” section on the installation page https://github.com/Field-Robotics-Lab/dave/wiki/Installation . I was not aware of this, and I just had the upstream repos, which gave various compiler errors.

I'm not done testing yet. There might be more, but I wanted to start the ticket in the interest of release timeline.

mabelzhang, 2021-07-27:
@bsb808 tagging in case you haven't seen

bsb808, 2021-07-27:
@mabelzhang - I'm playing catchup.  I'll work on this today and convert the tasks to issues with assignments.  Great suggestions!

M1chaelM, 2021-07-27:
@mabelzhang Just adding a note here that I was tasked with updating the installation instructions for the host-based and docker install. Currently these pages are in disarray, as you've noticed, but I should have new versions up soon. I think this will take care of your first two items.

M1chaelM, 2021-07-27:
Another note: if we are switching to VCS tool, this substantially changes the installation tutorials. To streamline a bit, can we resolve issue #115 before finalizing installation tutorials?

bsb808, 2021-07-27:
> Another note: if we are switching to VCS tool, this substantially changes the installation tutorials. To streamline a bit, can we resolve issue #115 before finalizing installation tutorials?

Yes - I will do so.

bsb808, 2021-07-27:
I converted all of these to individual issues with assignments.  I believe this links the issues so that we can divide and conquer.  If necessary, will close when the linked issues are addressed.

bsb808, 2021-07-30:
Closing b/c all the sub-issues (bullet points) have been successfully addressed.

closed 2021-07-30 by bsb808