Field-Robotics-Lab/dave #122: Update host install tutorials to sync with Docker images

M1chaelM, 2021-07-27:
# Context
* The [dockwater repository](https://github.com/Field-Robotics-Lab/dockwater) contains docker images intended to support CI testing, local development, and cloudsim environments for our various water-related simulation repositories.
* The Dockerfiles for these images give a clear description of the dependencies needed to build our environments.

# Problem
* Our host installation tutorials give instructions for installing needed dependencies directly on the host, but these are sometimes inconsistent, or sometimes take different approaches such that it is unclear whether they are equivalent.
* These are difficult to test without reinstalling the operating system.

# Solution
We should get the docker images working, **then** synchronize the host installation instructions to the docker images.

bsb808, 2021-07-27:
By "host installation tutorials" are you referring to source repositories that are cloned into the user's workspace?  If so, would your use of vcstool clear this up?  See https://github.com/Field-Robotics-Lab/dave/wiki/Clone-Dave-Repositories

M1chaelM, 2021-07-29:
This is addressed in the revised Host install tutorial: https://github.com/Field-Robotics-Lab/dave/wiki/Install-Directly-on-Host

closed 2021-07-29 by M1chaelM