Field-Robotics-Lab/dave #123: Noetic uuv_ws VCS output

crvogt, 2021-07-27:
repositories:
  src/dave:
    type: git
    url: https://github.com/field-robotics-lab/dave
    version: master
  src/ds_msgs:
    type: git
    url: https://bitbucket.org/whoidsl/ds_msgs.git
    version: master
  src/ds_sim:
    type: git
    url: https://bitbucket.org/whoidsl/ds_sim.git
    version: master
  src/eca_a9:
    type: git
    url: https://github.com/uuvsimulator/eca_a9.git
    version: master
  src/gtri_based_sonar:
    type: git
    url: https://github.com/Field-Robotics-Lab/gtri_based_sonar.git
    version: master
  src/hydrographic_msgs:
    type: git
    url: https://github.com/apl-ocean-engineering/hydrographic_msgs.git
    version: main
  src/nps_uw_multibeam_sonar:
    type: git
    url: https://github.com/field-robotics-lab/nps_uw_multibeam_sonar
    version: main
  src/nps_uw_sensors_gazebo:
    type: git
    url: https://github.com/Field-Robotics-Lab/nps_uw_sensors_gazebo.git
    version: master
  src/rexrov2:
    type: git
    url: https://github.com/uuvsimulator/rexrov2.git
    version: master
  src/uuv_manipulators:
    type: git
    url: https://github.com/field-robotics-lab/uuv_manipulators
    version: master
  src/uuv_simulator:
    type: git
    url: https://github.com/field-robotics-lab/uuv_simulator
    version: master

crvogt, 2021-07-27:
I'm planning to go through and trim extraneous branches again.

bsb808, 2021-07-27:
I don't believe we need  https://github.com/Field-Robotics-Lab/gtri_based_sonar.git anymore.
I'm removing it from the list.

bsb808, 2021-07-27:
Also, we shouldn't need  https://github.com/Field-Robotics-Lab/nps_uw_sensors_gazebo.git after the v3.1_reorg.

bsb808, 2021-07-29:
Closing this as all these repos have been folded into the vcs list.

closed 2021-07-29 by bsb808