Field-Robotics-Lab/dave #130: Investigate and resolve how to refer to python3 in Noetic

bsb808, 2021-07-28:
Should shebangs in Python source files be...
```
/usr/env python
```
OR
```
/usr/env python3
```

See http://wiki.ros.org/UsingPython3/SourceCodeChanges

We will work on this after release 4.0.0

bsb808, 2021-08-09:
## References
Read through See http://wiki.ros.org/UsingPython3/SourceCodeChanges and then the referenced [PEP-394: The "python" Command on Unix-Like Systems](https://www.python.org/dev/peps/pep-0394/).

## Recommendation

Use specific shebang for Python3 source files: `/usr/bin/env python3` for all ROS Noetic code.

## Justification

* We are targeting a ROS Noetic on Ubuntu 20.04
* ROS Noetic is exclusively Python 3
* Ubuntu 20.04 includes Python 3.x as the sole default (no Python 2.x)
* Ubuntu 20.04 does not provide a `python` command at all by default, but does provide a `python3` command by default.
* [PEP-394 - For Python runtime distributors](https://www.python.org/dev/peps/pep-0394/#for-python-runtime-distributors) encourages changing "less` specific shebangs to more specific ones."
* [PEP-394 - For Python script publishers](https://www.python.org/dev/peps/pep-0394/#for-python-script-publishers) recommends using  a specific shebang lines tailored to the target environment.
* " In Ubuntu, all python packages use explicit python3 or python2 interpreter and do not use unversioned /usr/bin/python at all" - see [python-is-python3](https://launchpad.net/ubuntu/focal/+package/python-is-python3)
* Using the generic `/usr/bin/env python` shebang has adverse effects:
    * For Python 3 exclusive systems, it requires a package such as   [python-is-python3](https://launchpad.net/ubuntu/focal/+package/python-is-python3), which adds a dependency and would likely break legacy Python2 scripts.
    * The best use-case for generic shebang is for code that is meant to support execution for either Python2 or Python3.  Our code is not meant to be executed by Python2.

bsb808, 2021-08-09:
@mabelzhang If you concur, I'll close the issue.

mabelzhang, 2021-08-13:
I've asked our Noetic + Python expert internally about this ticket specifically.

First, we still recommend keeping the shebangs as `python`. `catkin_install_python()` (which you are already using in
https://github.com/Field-Robotics-Lab/dave/blob/281e42b5393ebe26f9a31b63c5c595d3b8f5d89d/examples/dave_nodes/CMakeLists.txt#L41-L44
and that's the only package with Python scripts) will change the shebangs for you automatically. It does that in both the install space and the devel space (this project is using the devel space only, which is when you run `catkin_make` only and no install). So this setup should already work for you, in both Melodic and Noetic.

What I notice here is that only 2 out of 5 files are in that command, and you probably want to add the remaining 3 files - `dvl_state_and_gradient_dsl.py`, `dvl_state_and_gradient_uuvsim.py`, and `dvl_gradient_plot.py`.

Second, the other concern I have is the Melodic support, which we claim is community-supported. If we change the shebangs to `python3`, they will no longer work in Melodic. Melodic installs Python modules using Python 2, and if the scripts require Python 3, they won't be able to import any of the modules installed in Python 2.

If you decide to go with `python3` shebang, then in order to uphold that community support claim and help out the community, you'd either have another branch for Melodic, or put in the documentation that in Melodic, people would have to change the shebangs to `python`.

Third, my colleague pointed out that the adverse effects in the bullet list above are not true. You should not need a third-party package to execute those scripts.

What was the reason to look into using `python3` in the shebang? Do some files not work? I would recommend adding those 3 files mentioned above to `catkin_install_python()`.

M1chaelM, 2021-08-13:
@mabelzhang Thanks, this is helpful. I can only speak for myself, but what I thought we were doing was requiring our Melodic users to install Python3 as part of their environment, which didn't seem like a big deal. It sounds like that is not right because  "Melodic installs Python modules using Python 2", but I'm a little confused about that. Specifically, what does it mean for Melodic to install a Python module? 

Probably out of ignorance, I would typically do the installation of required modules without involving the ROS build system. I think we were also skipping the install process altogether for Python scripts called as nodes (which is why they are missing in that list) and just expecting them to function the same was as standalone programs. It sounds like the recommendation is to use the build system to manage these scripts and their dependencies.

That seems fine to me if that's the recommendation, although I don't fully understand what is being gained from this extra management layer. (I can see we get automatic shebang correction, but it seems the cost of this, and also the reason it's necessary, is that we are unable to use Python3 in Melodic.)

mabelzhang, 2021-08-14:
> what does it mean for Melodic to install a Python module?

I'm not an expert on the inner workings of dependencies in ROS. In practice, I have seen things like that happen, when a script can't find even the most basic Python packages, which I was sure were installed, but they were installed for the other Python version.

Under the hood, I think part of it has to do with how [rosdep keys](https://github.com/ros/rosdistro/blob/master/rosdep/python.yaml) are resolved. When you run `rosdep update`, rosdep finds system dependencies that the keys are mapped to. For Melodic, many packages probably list the Python 2 version of the rosdep keys, and rosdep would end up installing dependencies in Python 2. For example, in `rospy`, they specify the rosdep keys for Python 2 and 3 dependencies separately: https://github.com/ros/ros_comm/blob/melodic-devel/clients/rospy/package.xml

That's my educated guess. @sloretz can correct me.

> the cost of this, and also the reason it's necessary, is that we are unable to use Python3 in Melodic.)

I found some information here about how to get Python 3 to work with Melodic. The environment variable `ROS_PYTHON_VERSION` is used in the `rospy` example above.
https://answers.ros.org/question/295012/how-can-i-install-ros-melodic-with-python3/

These are the recommendations of doing things correctly. At the end of the day, the correct way may not be the most convenient. I see these as trade offs to being backward-compatible. Asking Melodic users to install Python 3 is one side of the trade off, and the other side is to maintain your Python scripts so that they can be run in both Python 2 and Python 3, in order to support both Melodic and Noetic.

Looking at the number of scripts in this repo, the latter shouldn't be a big obstacle. In everyday usage, there are only a handful of things that really require Python 3, especially if these scripts originally came from a Melodic environment. If you don't envision changing the scripts a lot, there shouldn't be too much overhead to do things the correct way. If you do change the scripts often, then running CI in Melodic using Python 2, and in Noetic using Python 3 would check that for you, though that's more overhead.

bsb808, 2021-09-02:
@mabelzhang, @M1chaelM  and @sloretz - This is great discussion and I've learned a lot thinking through this and following your expertise.

Here is my attempt at a summary of the key points:

1.  Advantage of generic python shebang is that, if all of our scripts are included in the `catkin_install_python` CMake directive, then Catkin will be smart about a specific shebang for Melodic/Python2 or Noetic/Python3.
2.  This advantage could, in theory, make maintenance simpler as the same scripts can be used for Melodic or Noetic support.

Here is my opinion, for our particular use-case (not for the general use-case).  

A key part of my personal use-case is that, when debugging, I tend to run Python scripts directly from the src directory and am not in the habit of using the  `catkin_install_python` directive ubiquitously.   I also tend to mix and match ROS with other libraries and infrastructure (e.g., LCM) and prefer to keep things as agnostic as possible, so relying on catkin build/make to run a python script correctly seems like overkill.

From the general, pure-ROS case, I can totally see why having all Python scripts run through catkin and used from devel/install makes sense.

* I stand by "adverse effects" in the original justification because, at this point, I've experienced in 2 or 3 projects outside of Dave.
* To me, relying on catkin and developer/maintainers to include 100% of python scripts in catkin to automatically determine which major version of Python to run is brittle and can lead to uncertainty in debugging, i.e., should I debug the copy in src, devel or install?
* Regardless of this issue, I think we will need to have a dedicated Melodic branch of some/all of our repos to be able to support both Melodic and Noetic. 
* Writing Python code that we are going to test with both Python2 and Python3 adds to our maintenance burden.  I'd prefer to move on to Noetic and only have to test with Python3.

At the end of the day, I think this is a bit of a stylistic question as there seems to be good reasons to go in both directions and no hard showstoppers for either choice.

If we are going to end up with separate branches for Melodic and Noetic (which I think we will), then, for the way I work, it seems simpler to be overt with the shebangs.

M1chaelM, 2021-09-02:
I will defer to the collective on this, but I thought I'd add a few comments that seem relevant to me:

* An advantage of installing scripts in catkin, I think, is that it will further simplify the Dockerfiles associated with the different operating systems in the dockwater repo. Right now I have to do extra things to make sure our Python scripts can find the dependencies they need. I'd have to check, but I suspect this will also be handled automatically if we let catkin do it.
* If I need to run a script directly for debugging and the shebang is not set up by default, I would personally just override it by calling `python3 <script>` instead. I don't do this often enough for it to be a real hassle, but if it were there's always the option to set the local "python" to point to "python3" or run from a python virtual environment if you don't want to mess with the global settings.
* I agree it is a burden to maintain compatibility with multiple versions of Ubuntu / ROS. However, writing Python that works in both 2 and 3 seems pretty easy. I can't think of something off the top of my head that works in Python3 but not Python2. Most of the problems go the other direction, but we already had to fix these anyway.
* I agree that I'm not sure whether having catkin manage Python scripts is very useful to me personally, since I'm accustomed to keeping track of Python dependencies myself. That said, this might just be a matter of getting used to something new (to me). In either case, the conversation about whether this is a good design seems to me to be a different question than the current one, which is should our code follow the convention that's currently in place. I'm a little hesitant to strike out on our own here because my view is that the advantages of adhering to almost _any_ standard outweigh the gains of almost any deviation.

Because of the above, I'd lean slightly toward adopting the generic python shebangs for this project for now and VRX, and separating out the discussion of whether `catkin_install_python` is the best way of doing things.

bsb808, 2021-12-15:
Going with @mabelzhang 's suggestion and adding the few Python programs to the `catkin_install_python` directive.  Will close with PR #175.

bsb808, 2021-12-16:
Closing with the merge of PR #175

closed 2021-12-16 by bsb808