Field-Robotics-Lab/dave #133: Ambiguous Tutorial - New Underwater Vehicle

crvogt, 2021-07-29:
https://github.com/Field-Robotics-Lab/dave/wiki/New-Underwater-Vehicle

The tutorial is meant to show a user how to "integrate a new robot ".

Following the tutorial, the reader would need to have enough knowledge that it renders the example irrelevant, suggesting more information is required. Alternatively, the tutorial becomes a "smilodon usage tutorial" or is removed and the smilodon example is found https://github.com/Field-Robotics-Lab/dave/wiki/vehicle_examples

Is the user meant to be following through the code?

bsb808, 2021-07-30:
https://github.com/Field-Robotics-Lab/dave/wiki/New-Underwater-Vehicle was intended to be a tutorial no now to integrate a new vehicle, i.e, if a user had their own custom ROV/AUV/UUV, this tutorial would explain how to get that new model into Dave.  The included Dave models are meant as general-purpose surrogate vehicles, with the expectation that many users have their own vehicle platfroms.

https://github.com/Field-Robotics-Lab/dave/wiki/vehicle_examples is intended as a list of the general-purpose surrogate vehicles included in dave.

bsb808, 2021-07-30:
Moving this to 4.0.1 to make the schedule.  Let's try to get this addressed ASAP.