Field-Robotics-Lab/dave #138: Move manipulation tutorials' images to wiki

mabelzhang, 2021-07-30:
(Post-4.0.0 release okay.)

There seem to be inconsistent locations where tutorials have their images.

Most of the manipulation tutorials have their images in the directory `tutorial_pictures` in the repo (or `extras/tutorial_pictures` in `v3.1_reorganization` branch, to be merged into `master` for 4.0.0 release).
Some manipulation tutorials have images associated with a user account, with URLs that look like `https://user-images.githubusercontent.com/13877149/91655629-f0356a00-ea7f-11ea-8cea-3feccb30f19f.png`

Looking at other tutorials in the wiki, they have images directly uploaded to the wiki, with URLs that look like `https://raw.githubusercontent.com/wiki/Field-Robotics-Lab/dave/screenshots/uuv_reference_view.png`

For consistency and easier maintenance, it's probably good to move these images to the wiki, like the other tutorials. That way we don't need to have the `extras/tutorial_pictures` directory that really only host images in the manipulation tutorials.

The 5 manipulation tutorials referenced above:
https://github.com/Field-Robotics-Lab/dave/wiki/Swapping-out-the-Oberon7-arm-with-another-manipulator
https://github.com/Field-Robotics-Lab/dave/wiki/Electrical-Plug-Mating-Plugin
https://github.com/Field-Robotics-Lab/dave/wiki/BOP-Panel-Manipulation-Mission
https://github.com/Field-Robotics-Lab/dave/wiki/Manipulator-Force-Feedback
https://github.com/Field-Robotics-Lab/dave/wiki/Teleop-Bar-Retrieval

bsb808, 2021-07-30:
Concur - post 4.0.0

bsb808, 2021-08-04:
Mabel will assign sections of repos.  Move all images to wiki repo.

mabelzhang, 2021-08-05:
Move to `images/` in wiki repo:

* [ ] `tutorial_pictures` in source code repo (10), `screenshots_fls_v1` (3), `screenshots_sonar_v2` (11) @mabelzhang
* [ ] `tutorials/images` (24) @dtdavi1 
* [ ] `screenshots` (19), `pictures` (1) @j-herman

(I'm taking this off @M1chaelM's plate since he's relieving me of chief warrior duty.)

Images in user spaces:
From `$ grep -rI "user-images.githubusercontent"`, there are 36 images across 8 wiki pages. It might be easiest if we each do the tutorials we reviewed for 4.0.0 release?
* [ ] I can take care of the ones in `Manipulator-Models.md` (1), `Electrical-Plug-Mating-Plugin.md` (4), `Swapping-out-the-Oberon7-arm-with-another-manipulator.md` (1) @mabelzhang 
* [ ] `Ocean_current_draft.md` (1), `Ocean-Current.md` (5) @j-herman do you want to do those?
* [ ] `Multibeam-Forward-Looking-Sonar.md` (21), `Install-on-Windows-using-WSL2.md` (2), `Bathymetry-Integration.md` (1) @woensug-choi do you want to do those?

Thanks everyone!

woensug-choi, 2021-08-13:
I've updated `Multibeam-Forward-Looking-Sonar.md`, `Install-on-Windows-using-WSL2.md`, `Bathymetry-Integration.md` on wiki repo and pushed them :) All images used in those documents are now in `images` directory.

mabelzhang, 2021-09-02:
Closing as all items in check list are completed. Thank you everyone!

closed 2021-09-02 by mabelzhang