Field-Robotics-Lab/dave #142: Outstanding tasks for manipulation tutorials

mabelzhang, 2021-07-30:
(Post-release 4.0.0 okay)

### Common to all 5 manipulation tutorials below
- [ ] Move associated files in `uuv_dave` out of `legacy/`
   - Launch files go to `examples/dave_demo_launch`?
   - Plugins go to `gazebo/.....`? Can't guess the right subdirectory
- [ ] Fix tf spam - I don't remember if these were resolved in other parts of the repo. Am I the only one getting these?
   ```
   [ WARN] [1627632533.852584126, 262.972000000]: TF_REPEATED_DATA ignoring data with redundant timestamp for frame rexrov/base_link_ned at time 262.972000 according to authority unknown_publisher
   Warning: TF_REPEATED_DATA ignoring data with redundant timestamp for frame rexrov/base_link_ned at time 262.972000 according to authority /gazebo
            at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
   ```

### Each tutorial specifically

https://github.com/Field-Robotics-Lab/dave/wiki/Swapping-out-the-Oberon7-arm-with-another-manipulator/
See above.

https://github.com/Field-Robotics-Lab/dave/wiki/Manipulator-Force-Feedback/
See above

https://github.com/Field-Robotics-Lab/dave/wiki/Electrical-Plug-Mating-Plugin/
- [ ] Move `uuv_mating`, in addition to associated files in `uuv_dave`, out of `legacy/`. Files are listed here https://github.com/Field-Robotics-Lab/dave/pull/40
- [ ] Tutorial doesn’t say how to actually move the robot. Joystick presumably. Should link to a page detailing joystick control for RexROV and the Oberon7.

https://github.com/Field-Robotics-Lab/dave/wiki/BOP-Panel-Manipulation-Mission	
- [ ] Tutorial doesn’t say how to actually move the robot. Joystick presumably. Should link to a page detailing joystick control for RexROV and the Oberon7.

https://github.com/Field-Robotics-Lab/dave/wiki/Teleop-Bar-Retrieval
This tutorial uses the same launch file (`uuv_dave.launch` I assumed and added) as the arm swapper tutorial, so should be set.
This tutorial has a bit more detail about the joystick. The two above should do the same, or they should all link to a page that talks about how to use the joystick. So far, I figured the left stick does z translation and rotation, and the right stick does x and y translation, for the base. I haven’t figured out how to move the actual arm.

bsb808, 2021-07-30:
@mabelzhang 

1. Ideally we'd get `uuv_mating` and `uuv_dave` bits and pieces moved out of the `legacy` prior to 4.0.0.  Can you do that today?
2. The tf spam issue seems larger - agree we'll need to put that to post 4.0.0

mabelzhang, 2021-07-30:
I can try to put them in the right places and others can check the locations in the PR.
I have a short meeting before noon PT and another thing due today, so I'll try.

bsb808, 2021-07-30:
I'm flip-flopping.  I want to make the release deadline that is 16 minutes away, so let's push this to 4.0.1

mabelzhang, 2021-07-30:
Okay. Yeah I'm getting crashes about TCP errors after moving `uuv_mating`. It's not gonna be done in the next 10 minutes.

mabelzhang, 2021-08-14:
I've addressed everything in this ticket in #149 and #150, other than the tf spam, which isn't specific to manipulation tutorials and could be tracked either here or in a separate issue.

mabelzhang, 2021-09-04:
Closing as all have been merged

closed 2021-09-04 by mabelzhang