Field-Robotics-Lab/dave #147: Use <author> tag in package.xml

mabelzhang, 2021-08-10:
During the recent repo reorganization for release 4.0.0, I noticed that the <author> tag is not used in `package.xml` files. Something to consider. Even though `git blame` shows the author per-line, commit histories for files are [not preserved when files are moved](https://stackoverflow.com/questions/2314652/is-it-possible-to-move-rename-files-in-git-and-maintain-their-history) (run `git log` on moved files, and you'll see all the history is lost).

For open source projects, arguably, it's important to credit the original authors. We don't do that in Gazebo / Ignition as there's no package.xml, but in ROS, we often have multiple <author> and multiple <maintainer> tags per package, for example, https://github.com/ros/robot_state_publisher/blob/noetic-devel/package.xml .

(We even automatically append authors of PRs for each release to the CHANGELOGs ([example](https://github.com/ros/robot_state_publisher/blob/noetic-devel/CHANGELOG.rst)) via Bloom, but Bloom is for releasing packages into rosdistro and not applicable here.)

Since the DAVE repo seems to have different contributors with their own expertise, and some have left the project, this is something to think about.

bsb808, 2021-12-02:
Proposed authors: https://docs.google.com/spreadsheets/d/1RsEDT80HVvjFNxyU4ATySshFTjZSPKICIs6GfQ18HQA/edit?usp=sharing

bsb808, 2021-12-16:
Addressed with PR #173

closed 2021-12-16 by bsb808