Field-Robotics-Lab/dave #148: catkin_make vs. catkin build

mabelzhang, 2021-08-10:
Consider using `catkin build` for compiling packages.

`catkin build` has been a more modern version to `catkin_make` for a few years.

The only additional dependency is `sudo apt-get install python-catkin-tools`.
The main difference is that `catkin build` builds each package in isolation, which is explained more in the first paragraph here https://catkin-tools.readthedocs.io/en/latest/migration.html

From a user experience point of view, `catkin build` produces much cleaner console output, and when you want to build one package, it really builds just that one package, instead of building a huge chain of other packages too. The former saves a lot of development time.

Typically, unless you are using some advanced build features, there isn’t much migration to be done, but the link above also has a migration guide.

mabelzhang, 2021-08-14:
Just a quick update. I've tried this in the dockwater Docker image and it compiles out of the box:
```
sudo apt-get update
sudo apt-get install python3-catkin-tools
catkin build
```

Give it a try.

After building once, try building a single package (maybe change something somewhere, not sure if that's needed):
```
catkin_make --pkg dave_object_models
```
vs
```
catkin build dave_object_models
```
You'll notice that the former still goes and builds many other packages, whereas the latter only builds that specified package and nothing else.

woensug-choi, 2021-08-20:
Tested on Host installation case. `catkin build` works without any modifications. It compiles fast since it generically utilizes all cores of the system to compile in parallel.

mabelzhang, 2021-09-02:
Thanks for trying it out!

To move forward on this ticket, it's just a matter of updating the installation tutorials, if people want to adopt `catkin build`.

woensug-choi, 2021-09-07:
I vote to update to `catkin build` ! It looks much better. If we are going to do this, we need to try to install everything again from the scratch on a clean machine to check everything is there. My log also has `pip3 install catkin_pkg`.

mabelzhang, 2021-11-04:
Created a PR in `dockwater` to address this https://github.com/Field-Robotics-Lab/dockwater/pull/11
That should give us a default Docker environment that can run `catkin build`.
I've tested this only with DAVE and not other repos that use dockwater.

M1chaelM, 2021-11-05:
@bsb808 I tested this on dockwater last night and left a note there, but just so it doesn't get overlooked: running
```
pip3 install osrf-pycommon
```
on a freshly built noetic container gives the error:
```
Requirement already satisfied: osrf-pycommon in /usr/lib/python3/dist-packages (1.0.0)
```
so I'm wondering whether this issue was resolved upstream. The dependency is supposed to be installed with the `python3-catkin-tools` package, which is already included in the image.

M1chaelM, 2021-11-05:
@mabelzhang I'm having difficulty reproducing the "out of the box" experience with catkin build. Running through the install instructions on our wiki with `catkin_make` works fine, but if I delete everything in the workspace and try to run with `catkin build` I see tons of errors, ending in this grim summary:
```
[build] Summary: 12 of 46 packages succeeded.                                                                                              
[build]   Ignored:   None.                                                                                                                 
[build]   Warnings:  None.                                                                                                                 
[build]   Abandoned: 24 packages were abandoned.                                                                                           
[build]   Failed:    10 packages failed.
```
The first error in the output is:
```
Errors     << dave_sensor_models:cmake /home/mrmccarr@ern.nps.edu/uuv_ws/logs/dave_sensor_models/build.cmake.000.log                       
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by
  "uuv_sensor_ros_plugins" with any of the following names:

    uuv_sensor_ros_pluginsConfig.cmake
    uuv_sensor_ros_plugins-config.cmake

  Add the installation prefix of "uuv_sensor_ros_plugins" to
  CMAKE_PREFIX_PATH or set "uuv_sensor_ros_plugins_DIR" to a directory
  containing one of the above files.  If "uuv_sensor_ros_plugins" provides a
  separate development package or SDK, be sure it has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:11 (find_package)

cd /home/mrmccarr@ern.nps.edu/uuv_ws/build/dave_sensor_models; catkin build --get-env dave_sensor_models | catkin env -si  /usr/bin/cmake /home/mrmccarr@ern.nps.edu/uuv_ws/src/dave/models/dave_sensor_models --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/mrmccarr@ern.nps.edu/uuv_ws/devel/.private/dave_sensor_models -DCMAKE_INSTALL_PREFIX=/home/mrmccarr@ern.nps.edu/uuv_ws/install; cd -
```
Any idea why I would be seeing this but not you or @woensug-choi ?

mabelzhang, 2021-11-06:
> left a note there

Just replied to the dockwater PR.

> tons of errors, ending in this grim summary:

Not sure. My `rospack` output says
```
$ rospack find uuv_sensor_ros_plugins
/home/master/dave_ws/src/uuv_simulator/uuv_sensor_plugins/uuv_sensor_ros_plugins
```

I have these in `uuv_simulator`:
```
$ git remote -v
origin	git@github.com:Field-Robotics-Lab/uuv_simulator.git (fetch)
origin	git@github.com:Field-Robotics-Lab/uuv_simulator.git (push)
$ git branch
* master
```

Maybe try removing the generated `CMakeLists.txt` in `src` too? I don't know if that would help though.

M1chaelM, 2021-11-18:
I'm still getting errors even when I delete the src and start from scratch. To try to ensure a clean environment, I ran the dockwater image repository without my home directory mounted. Then I configured my ssh-agent, and following the installation instructions on our [Wiki] (https://github.com/Field-Robotics-Lab/dave/wiki/Clone-Dave-Repositories) I ran:
```
source /opt/ros/noetic/setup.bash
mkdir -p uuv_ws/src
cd uuv_ws/src
git clone https://github.com/Field-Robotics-Lab/dave.git
vcs import --skip-existing --input dave/extras/repos/dave_sim.repos .
cd ../
catkin build
```
But this still fails. Also, running 
```
pip3 install osrf-pycommon
```
Still produces:
```
Requirement already satisfied: osrf-pycommon in /usr/lib/python3/dist-packages (1.0.0)
```
I'm really perplexed because we're using the same image. I'll start looking through the individual errors and warnings now.

M1chaelM, 2021-11-18:
Warnings:
```
catkin_package() include dir  '/home/mrmccarr@@ern.nps.edu/uuv_ws/build/dave_gazebo_world_plugins' should be placed in the devel space instead of the build space
```
```
catkin_package() include dir  '/home/mrmccarr@@ern.nps.edu/uuv_ws/build/uuv_gazebo_plugins' should be placed in the devel space instead of the build space
```
```
StratifiedCurrentVelocity.proto: warning: Import any.proto but not used.
```
```
the rosdep view is empty: call 'sudo rosdep init' and 'rosdep update'
```
```
The dependency target "dave_gazebo_world_plugins_msgs" of target "dsros_ros_dvl" does not exist. This warning is for project developers.  Use -Wno-dev to suppress it.
```
```
[libprotobuf WARNING google/protobuf/compiler/parser.cc:562] No syntax specified for the proto file: Double.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
[libprotobuf WARNING google/protobuf/compiler/parser.cc:562] No syntax specified for the proto file: Accel.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
```
```
/home/mrmccarr@ern.nps.edu/uuv_ws/src/uuv_simulator/uuv_sensor_plugins/uuv_sensor_ros_plugins/src/RPTROSPlugin.cc: In member function ‘virtual bool gazebo::RPTROSPlugin::OnUpdate(const gazebo::common::UpdateInfo&)’:
/home/mrmccarr@ern.nps.edu/uuv_ws/src/uuv_simulator/uuv_sensor_plugins/uuv_sensor_ros_plugins/src/RPTROSPlugin.cc:119:1: warning: control reaches end of non-void function [-Wreturn-type]
  119 | }
      | ^
```
Errors:
```
[libprotobuf WARNING google/protobuf/compiler/parser.cc:562] No syntax specified for the proto file: SensorDepth.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
/usr/bin/ld: cannot find -ldave_gazebo_world_plugins_msgs
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/dsros_sensors.dir/build.make:197: libdsros_sensors.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:3206: CMakeFiles/dsros_sensors.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
```

dtdavi1, 2021-11-19:
@M1chaelM @mabelzhang I get the same error that Michael is getting.  I tried "catkin build dave_gazebo_world_plugins" and then "catkin build" to force it to build the plugins package first (shouldn't matter, but I was grasping at straws) with no success.  Interestingly, the dave_gazebo_world_plugins build works but gives the following warning:

CMake Warning at /opt/ros/noetic/share/catkin/cmake/catkin_package.cmake:418 (message):
  catkin_package() include dir
  '/home/dtdavis/uuv_ws/build/dave_gazebo_world_plugins' should be placed in
  the devel space instead of the build space
Call Stack (most recent call first):
  /opt/ros/noetic/share/catkin/cmake/catkin_package.cmake:102 (_catkin_package)
  CMakeLists.txt:46 (catkin_package)

Is there something missing from CMakeLists.txt that makes it install someplace that catkin build isn't finding it?  When I ran "catkin build", I noticed that the same warning occurs with a few of the other packages as well (before it abandons the build for 37 packages).

mabelzhang, 2021-11-19:
I just created a new directory for a new workspace and ran the commands above in https://github.com/Field-Robotics-Lab/dave/issues/148#issuecomment-973327531

I was able to reproduce the error:
```
Errors     << ds_sim:make /home/master/dave_tmp_ws/logs/ds_sim/build.make.000.log
[libprotobuf WARNING google/protobuf/compiler/parser.cc:562] No syntax specified for the proto file: SensorDepth.proto. Please use 'syntax = "proto2";' or 'syntax = "proto3";' to specify a syntax version. (Defaulted to proto2 syntax.)
/usr/bin/ld: cannot find -ldave_gazebo_world_plugins_msgs
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/dsros_sensors.dir/build.make:197: libdsros_sensors.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:3206: CMakeFiles/dsros_sensors.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
cd /home/master/dave_tmp_ws/build/ds_sim; catkin build --get-env ds_sim | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

...............................................................................
Failed     << ds_sim:make                                             [ Exited with code 2 ]
Failed    <<< ds_sim                                                  [ 1 minute and 14.4 seconds ]
```

I haven't seen it before.
It's weird that it's failing on `ds_sim` though, because the error seems to be this line:
```
/usr/bin/ld: cannot find -ldave_gazebo_world_plugins_msgs
```

That library is the protobuf message in `dave_gazebo_world_plugins`.
Looking at the CMakeLists.txt, I believe that library is supposed to built first, before code that depend on the protobuf library is built.
Since the only code that uses the protobuf library is in the package itself, the error could mean that the library is built out of order.

In any case, it's weird that it worked a few months before and not now. Maybe something changed in catkin tools.

mabelzhang, 2021-11-19:
Oh I know why I didn't get this in my regular workspace. My hash of ds_sim is from July, and it doesn't have a dependency on `dave_gazebo_world_plugins_msgs`.

In the new workspace with the latest code, that dependency is there.

Probably has to do with the target being exported and depended on appropriately.

M1chaelM, 2021-12-02:
After considerable negotiation, next steps on this issue have been dumped on @mabelzhang, though we agree the source of the error is code from @dtdavi1 , who will support refactoring efforts as necessary.

mabelzhang, 2021-12-15:
The PR in ds_sim (linked immediately above this comment) should fix it. Then we should be able to use `catkin build`.

woensug-choi, 2021-12-15:
Hurray it works now :D

mabelzhang, 2021-12-16:
Closing via https://github.com/Field-Robotics-Lab/ds_sim/pull/5

mabelzhang, 2021-12-16:
Oh, actually, I just realized this can't be closed yet.
`dockwater` still needs to be updated, and the instructions in the wiki for installation accordingly.

woensug-choi, 2021-12-23:
It compiles ok. But I get a related error msg on the runtime. Hmm...

https://github.com/Field-Robotics-Lab/ds_sim/pull/5#issuecomment-1000125169

woensug-choi, 2022-01-18:
With,  https://github.com/Field-Robotics-Lab/ds_sim/pull/6 and https://github.com/Field-Robotics-Lab/dave/pull/179
Yes! It works now :D https://github.com/Field-Robotics-Lab/ds_sim/pull/5#issuecomment-1000125169 is solved! 

Now nothing is stopping us from migrating to `catkin build` 👍

mabelzhang, 2022-01-26:
Some more related fixes to CMake dependencies setup were recently merged:
https://github.com/Field-Robotics-Lab/ds_sim/pull/6: Export libraries for downstream use
https://github.com/Field-Robotics-Lab/dave/pull/182: Fixes protobuf header installation location so that the same source code works for both install and no-install.

Next steps: If @M1chaelM validates things work, this [wiki page](https://github.com/Field-Robotics-Lab/dave/wiki/Build-Dave-Environment) can be updated to use `catkin build`.
Is that the only place?

woensug-choi, 2022-01-27:
[image]

Search result on dave.wiki repo.

mabelzhang, 2022-01-27:
I thought about it a bit more. Because there were some problems, likely due to `catkin bulid` being more strict with correctness, I'm leaning toward listing `catkin build` as an experimental alternative, rather than replacing `catkin_make`. That way, people can use it as a choice, and if there are more problems, we can fix them as they come up.

I'm also leaning toward only doing that on the main installation page, since it's experimental. That would be `Build-Dave-Environment.md`, `dave-with-Docker.md`, `install-Directly-on-Host-(Noetic).md`, `noetic_docker.md` from the list above. Actually, I did a `grep` and found that some of these pages aren't linked from anywhere. Probably some refactoring had happened, and these pages haven't been deleted. If I click on the Installation page https://github.com/Field-Robotics-Lab/dave/wiki/Installation , all the pages now link to `Build-Dave-Environment`. So that might be the only page we need to add `catkin build` as an alternative.

mabelzhang, 2022-01-29:
@M1chaelM Thoughts and preferences on the very last comment?

M1chaelM, 2022-02-25:
@mabelzhang Sorry, finally getting back to this. `catkin build` now works for me so I'm fine either way. Happy to leave it as an alternative until we fix those warnings (I get 4 of them).

mabelzhang, 2022-02-26:
Thanks!

These warnings? https://github.com/Field-Robotics-Lab/dave/issues/148#issuecomment-973342059
Or some other warning?

I updated this wiki page with an experimental section https://github.com/Field-Robotics-Lab/dave/wiki/Build-Dave-Environment

mabelzhang, 2022-03-18:
Closing since wiki already updated.
Feel free to reopen when we have bandwidth to look at the warnings again.

closed 2022-03-18 by mabelzhang