Field-Robotics-Lab/dave #151: Clarification of urdf/ vs models/

mabelzhang, 2021-08-14:
I ran into a confusion and thought I'd bring it up. This is mostly a question for @dtdavi1 of what's intended to go into these two directories for future reference.

I was looking at the directory contents to decide where to move models in `legacy/uuv_dave` to (turned out they were already moved), and I got really confused.

There is `urdf/robots` and `models/dave_robot_models`.
There is `urdf/sensors` and `models/dave_sensor_models`.

There seems to be some duplication. For example, some of them have the same mesh files
```
$ ls models/dave_sensor_models/models/nortek_dvl1000_300/meshes/ -1
DVL1000-300m.dae
tx-DVL1000-300m.png

$ ls urdf/sensors/nortek_dvl1000_300_dsl_description/meshes/ -1
DVL1000-300m.dae
tx-DVL1000-300m.png
```

Initially, judging from the names, I thought maybe `urdf` contains just the URDF/xacro files, and `models` contain meshes, or the other way around.
At the end, I'm now of the conclusion that `urdf/` contains models from DSL and uuv_sim, whereas `models/` contains models for DAVE, and there might be slight differences?
The two names kind of mean the same thing. It would be good to document this somewhere or make it less confusing somehow.

dtdavi1, 2021-08-14:
@mabelzhang The dave_robot_models and dave_sensor_models packages contain SDF models for some (most?) of the robots and sensors that we've worked with (there's also a dave_object_models package with some of the miscellaneous objects that we've used in different worlds).  The urdf directory contains ROS robot and sensor description packages with XACRO templates for most of the same robots and sensors.  The models packages provide models that can be incorporated directly into Gazebo worlds, while the description packages have templates that can be used to dynamically spawn sensor-equipped robots in arbitrary worlds using the ROS spawn_model node.  The meshes are included in both versions to avoid cross-dependencies (i.e., an end user could use either form independently without having to rely on the other).

The DVL models have UUV Simulator and ds_sim versions, so their naming reflects the simulator version as well.  Alternatively, we could put both URDF versions in the same package if that's what everyone prefers.  If so, I can take care of that easily enough.  The SDF models will still need to be differentiated, though.

Most of the models in the legacy directory have already been moved to the new organization, but are duplicated in the legacy directory for now so that those examples will still work.  Once we're confident that everything we want to maintain is in the main section (i.e., when Brian says so), we'll delete the legacy directory completely.

dtdavi1, 2021-11-08:
@bsb808 I can go either way with this, so I want to verify the group (or community) preference before changing anything.  If the desire is to only have one copy of each mesh in the entire repo, I can update the URDF packages to refer to the meshes in SDF model directories.  If the desire is to have the packages as self contained as possible, then I think they're ok as is.  The demos will be unchanged regardless, so this can be resolved and merged back into the master branch transparently.

dtdavi1, 2021-12-16:
Meshes removed from URDF (description) packages and replaced with references to the model meshes where possible.  Merged into master with [#169](https://github.com/Field-Robotics-Lab/dave/pull/169).