Field-Robotics-Lab/dave #152: dave_demo.launch warming spams

woensug-choi, 2021-08-20:
There were `TF_REPEATED_DATA` spam on the console when running `roslaunch dave_demo_launch dave_demo.launch`. The spam can be eliminated by `2> >(grep -v TF_REPEATED_DATA)` at the end of the launching command as  `roslaunch dave_demo_launch dave_demo.launch 2> >(grep -v TF_REPEATED_DATA)`. But still, there were `at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp` spamming for dave_demo.launch.
```
[INFO] [1629440622.469918, 0.160000]: Started controllers: /rexrov/oberon7/azimuth/controller, /rexrov/oberon7/shoulder/controller, /rexrov/oberon7/elbow/controller, /rexrov/oberon7/roll/controller, /rexrov/oberon7/pitch/controller, /rexrov/oberon7/wrist/controller
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
         at line 278 in /tmp/binarydeb/ros-noetic-tf2-0.7.5/src/buffer_core.cpp
```

`2>/dev/null` instead of `2> >(grep -v TF_REPEATED_DATA)` would not produce any spam-like warnings.

Should we update our wiki for this?

mabelzhang, 2021-09-02:
Copying the relevant link from the Trello card here, for visibility and future reference.

My previous summary about this (bullet 1)
https://github.com/tfoote/nps_me4823/issues/14#issuecomment-830593917
which links to the original upstream discussions from `robot_localization` and `tf` and the `/de/null` workarounds here
https://answers.ros.org/question/359968/tf_repeated_data-error-with-robot_localization-ukf-package/
https://github.com/ros/geometry2/issues/467#issuecomment-695052790

woensug-choi, 2021-09-07:
https://github.com/Field-Robotics-Lab/uuv_simulator/pull/3 solved this issue.

closed 2021-09-07 by woensug-choi