Field-Robotics-Lab/dave #162: Docker vcs import with SSH key

dtdavi1, 2021-10-30:
vcs import from within the noetic docker image does not work with an SSH key.  It repeatedly asks for the passphrase for the key in the .ssh directory but does not accept the correct passphrase.  The git clones work correctly if they are done manually one at a time.

woensug-choi, 2021-11-03:
Would changing vcs script address from `git@github.com:Field-Robotics-Lab/dave.git` to `https://github.com/Field-Robotics-Lab/dave.git` will solve this problem?

dtdavi1, 2021-11-03:
Changing to the https address would sidestep the problem by using HTTPS instead of SSH for identity management and access control.  That's probably ok for users since they're public repositories.  Developers might find it annoying since they'd be required to enter their username and password to push commits back to the server (o manually change the remote addresses back to the git versions on their development machines).

woensug-choi, 2021-12-22:
https://stackoverflow.com/questions/23391839/clone-private-git-repo-with-dockerfile

dtdavi1, 2021-12-28:
I've looked into this some and concluded that the original issue results from a VCS quirk.  Because of the way that it's threaded, anything that requires user input wont work without a command-line flag that forces the whole VCS process to run in a single thread.  In addition to slowing things down by ditching multi-threading, it also requires the user to enter their SSH key password for every single download.  There's another potential workaround is to unlock the key for the Docker session (automating this is possible but a little clunky).  We can talk about which (if any) approach we want to take at out next meeting.  I think that @woensug-choi's post is describing is a slightly different manifestation of the larger VCS issue--forcing it to run as a single thread would probably fix what he's seeing, but automatically unlocking the SSH key would not.

woensug-choi, 2021-12-29:
How about mounting ssh key when running the `dockwater` image? This assumes that the host machine has a working ssh key to access github.
```
-v ~/.ssh:/home/tf_docker/.ssh:ro
```

dtdavi1, 2021-12-29:
@woensug-choi The directory containing the key is mounted already, but it still needs to be unlocked when the session starts.  It's easy enough to do, but the command differs slightly depending on whether the user's key is RSA or ECC (and as you note, it assumes they have a key in the first place).