Field-Robotics-Lab/dave #165: Sonar Simulator (Sonar Image in Cartesian Coordinates)?

loguna123, 2021-11-05:
Hi,

May I ask if the "/blueview_p900/sonar_image", is it in Cartesian Coordinates?
It somehow doesn't seem to be in polar form to me. 

BTW, nice work on the simulator, I enjoy it very much.

woensug-choi, 2021-11-11:
@loguna123 Sorry for the late notice of this issue!

The msg format of the `sonar_image` is in Cartesian Coordinates (`sensor_msgs::Image` format) mapped using OpenCV's `cv::ellipse` function. Which is lively shown in rqt_plot when launched with the templates.

If you need polar coordinates or something similar, use `sonar_image_raw` msg which is [UW APL's SonarImage](https://github.com/apl-ocean-engineering/hydrographic_msgs/blob/main/acoustic_msgs/msg/SonarImage.msg) format. The msg contains a linearized dataset (azimuth/elevation/range/intensities separately) which you may convert into any coordinates. It is much straight forward to plot into polar coordinates.

woensug-choi, 2021-12-07:
Closing this issue has been without any response for 20 days. I hope things worked out for you. If not, comment here, and I will revive this.

closed 2021-12-07 by woensug-choi