Field-Robotics-Lab/dave #170: Gazebo Protobuf Errors

ckeil1, 2021-12-06:
I am trying to run some of the manipulation and ocean current demos, and am running into errors that crash gazebo on launch. I see the following error in the launch spew:
```
[libprotobuf ERROR google/protobuf/descriptor_database.cc:58] File already exists in database: vector3d.proto
[libprotobuf FATAL google/protobuf/descriptor.cc:1358] CHECK failed: GeneratedDatabase()->Add(encoded_file_descriptor, size): 
terminate called after throwing an instance of 'google::protobuf::FatalException'
  what():  CHECK failed: GeneratedDatabase()->Add(encoded_file_descriptor, size): 
Service call failed: transport error completing service call: unable to receive data from sender, check sender's logs for details
[ERROR] [1638821274.641575, 0.000000]: Spawn service failed. Exiting.
```
These appear to be related to some of the custom plugins. The standard dave_demo.launch works fine, but several of the other demos have issues. Specifically, I have see this problem when running `roslaunch dave_demo_launch dave_transient_current_demo.launch`, and `roslaunch dave_demo_launch dave_plug_and_socket_demo.launch`, `roslaunch dave_demo_launch dave_electrical_mate.launch`. The launch error for the transient_current_demo looks like this:

```[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.110.113.105
[ INFO] [1638825209.917677682]: Finished loading Gazebo ROS API Plugin.
[ INFO] [1638825209.918630091]: waitForService: Service [/gazebo_gui/set_physics_properties] has not been advertised, waiting...
[Msg] Waiting for master.
[Msg] Connected to gazebo master @ http://127.0.0.1:11345
[Msg] Publicized address: 10.110.113.105
Warning [parser.cc:723] Can not find the XML attribute 'version' in sdf XML tag for model: NED frame. Please specify the SDF protocol supported in the model configuration file. The first sdf tag in the config file will be used 
[Msg] Loading world file [/home/colin/rum/src/dave/models/dave_worlds/worlds/dave_ocean_waves_transient_current.world]
[ INFO] [1638825210.362134501]: waitForService: Service [/gazebo/set_physics_properties] is now available.
[ INFO] [1638825210.376679281]: Physics dynamic reconfigure ready.
[Err] [InsertModelWidget.cc:402] Missing model.config for model "/home/colin/rum/src/tracbot_gazebo/models/lab"
[INFO] [1638825210.621682, 0.000000]: Calling service /gazebo/spawn_urdf_model
[libprotobuf ERROR google/protobuf/descriptor_database.cc:58] File already exists in database: vector3d.proto
[libprotobuf FATAL google/protobuf/descriptor.cc:1358] CHECK failed: GeneratedDatabase()->Add(encoded_file_descriptor, size): 
terminate called after throwing an instance of 'google::protobuf::FatalException'
  what():  CHECK failed: GeneratedDatabase()->Add(encoded_file_descriptor, size): 
[ERROR] [1638825210.949000, 0.000000]: Spawn service failed. Exiting.
Service call failed: transport error completing service call: unable to receive data from sender, check sender's logs for details
[Err] [Connection.cc:546] Connection[16] Closed during Read
[Err] [TransportIface.cc:385] Unable to read from master
[Err] [GLWidget.cc:900] Unable to connect to a running Gazebo master.
Aborted (core dumped)
Traceback (most recent call last):
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_service.py", line 523, in call
    responses = transport.receive_once()
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", line 744, in receive_once
    self.stat_bytes += recv_buff(sock, b, p.buff_size)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_base.py", line 111, in recv_buff
    raise TransportTerminated("unable to receive data from sender, check sender's logs for details")
rospy.exceptions.TransportTerminated: unable to receive data from sender, check sender's logs for details

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/colin/rum/devel/lib/uuv_assistants/publish_world_models.py", line 15, in <module>
    exec(compile(fh.read(), python_script, 'exec'), context)
  File "/home/colin/rum/src/uuv_simulator/uuv_assistants/scripts/publish_world_models.py", line 156, in <module>
    world_pub = WorldPublisher()
  File "/home/colin/rum/src/uuv_simulator/uuv_assistants/scripts/publish_world_models.py", line 41, in __init__
    self.add_meshes(meshes)
  File "/home/colin/rum/src/uuv_simulator/uuv_assistants/scripts/publish_world_models.py", line 79, in add_meshes
    prop = self._get_model_state(model_name, '')
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_service.py", line 442, in __call__
    return self.call(*args, **kwds)
  File "/opt/ros/noetic/lib/python3/dist-packages/rospy/impl/tcpros_service.py", line 533, in call
    raise ServiceException("transport error completing service call: %s"%(str(e)))
rospy.service.ServiceException: transport error completing service call: unable to receive data from sender, check sender's logs for details
```

dtdavi1, 2021-12-17:
@ckeil1 I'm not able to duplicate your bug, so I'll need more information to figure out why you're getting the errors.  Are you running in a Docker image (i.e., the one that you get from following the tutorial) or directly on the host?  What version of ROS and Gazebo are you using (we only actively support ROS Noetic and Gazebo 11)?  Etc.  The full Gazebo log output might also be helpful.

In the meantime, I would suggest pulling the current versions of the dave and ds_sim repositories and rebuilding everything with either "catkin build" (preferred) or "catkin_make".  You might want to clear out the build and devel directories of your ROS workspace to make sure it rebuilds everything.

mabelzhang, 2023-10-31:
Closing stale old ticket

closed 2023-10-31 by mabelzhang