Field-Robotics-Lab/dave #171: Multibeam Forward Looking Sonar demo noise

MariusDgr, 2021-12-12:
Hello, I have been trying to run the blueview p900 simulation with noise, as it is shown in the wiki page for the forward looking sonar. In the gif in the documentation showing the rqt_image_view window I can see that the image is continuously updating due to simulated noise, but when I try to run the demo my image is static. How can I replicate what is observed in the wiki page? (I want to at least also add the noise in the simulation).

I am trying to run the demo with the setup mentioned in the installation tutorial as legacy mode. Maybe this is the issue?

woensug-choi, 2021-12-13:
Hi, @MariusDgr Thank you for trying the plugin!

We've depreciated the sparkling noise (continuous change of random noise values even when the sonar scene is static). It seems in the real world, the sparkling noise does not occur unless the vehicle the sonar is attached is moved or the object in the scene is changed. These physical characteristics are exploited in some scenarios to recognize the changes in the scene by detecting those noise changes. The Gaussian noise value (the random noise value) is changed whenever the maximum distance of the objects in the scene is changed. I should leave a note on the wiki.

I've made a new tag called `artificalVehicleVibration`. It's defined as false as default. If you turn this on, you will be able to see those sparkling noise sonar image again.

MariusDgr, 2021-12-13:
I see, thank you. I will try running the demo again with this setting turned on.
Thank you for the fast response and good job with this project, it looks great!

woensug-choi, 2022-01-04:
No activity on this issue. If the problem persists, please reopen this issue.

closed 2022-01-04 by woensug-choi