Field-Robotics-Lab/dave #187: Mesh statistics for multibeam sonar response to degraded objects

mabelzhang, 2022-02-05:
Original conversation thread started in https://github.com/Field-Robotics-Lab/dave/pull/185#issuecomment-1025328474

Paraphrasing; @woensug-choi please edit if inaccurate.

We'd like to find in Blender or manually calculate some kind of mesh statistics from the degraded object, to be used to inform the multibeam sonar plugin, so that it returns signals that capture a subset of 1. micro-roughness, 2. geometric change, and 3. surface material change.

The reason is that sonar readings are affected by not only the geometric change in the mesh shape, but a number of other factors.

mabelzhang, 2022-02-05:
Replying to https://github.com/Field-Robotics-Lab/dave/pull/185#issuecomment-1029505079 here, to decouple this discussion from that PR.

> Or maybe we could ditch `subdivision` function called to the whole object and use a manual loop for each mesh element and give custom subdivision density. Maybe https://docs.blender.org/api/current/bpy.types.Mesh.html#bpy.types.Mesh.split_faces?

I suspect doing it manually will be difficult, because you might end up with holes in the mesh. My guess is that it probably takes deeper understanding of the mesh faces to do it right.

`split_faces` I think does something similar to this https://docs.blender.org/manual/en/latest/modeling/meshes/editing/mesh/split.html
Not exactly the same. The ones in the Mesh > Split menu are [`bpy.ops.mesh.edge_split`](https://docs.blender.org/api/current/bpy.ops.mesh.html#bpy.ops.mesh.edge_split), but I think that's what they mean by splitting.

mabelzhang, 2022-02-05:
> I think `calculate_normals` does not output anything. It's internal function to calculate normals.

Maybe not that one, but there has to be an API somewhere, unless they decided not to expose it - I can't think why they wouldn't expose it.

The GUI has data you want. Take a look below. "Edge Length," "Edge Angle," and normals for vertices, edges, and faces.
It's just a matter of finding the actual data in the API, not the checkboxes.
[image]

Will keep looking.

mabelzhang, 2022-02-05:
I think this is the API, just playing with tabbing for auto-complete in the Blender scripting prompt.

With a cylinder added, these are vertex normals:
```
>>> import bpy
>>> me = bpy.context.object.data
>>> len(me.vertices)
64
>>> me.vertices[0].normal
Vector((0.09799493104219437, 0.9951781034469604, 0.0))
```

Face normals:
```
>>> len(me.polygons)
34
>>> me.polygons[0].normal
Vector((0.09801729023456573, 0.9951847195625305, 0.0))
```

Split normals (from [here](https://blender.stackexchange.com/questions/240587/calculating-split-normals-using-python)):
Split normals are the magenta lines above, they are "vertex-per-face normals", and there are multiple per vertex.
```
>>> me.calc_normals_split()
>>> import numpy as np
>>> norms = np.empty(3 * len(me.loops))
>>> me.loops.foreach_get("normal", norms)
>>> nx, ny, nz = norms.reshape((-1, 3)).T
```

Which type of normals are you interested in, of the vertices, splits, or faces?

woensug-choi, 2022-02-08:
Hmm... still brainstorming how to proceed things forward. I am not sure whether we need this. The concept of using the deviation of the normals is assuming that we know the original normal. But as the sonar plugin is only dealing with the already degraded object in the world, that information is unavailable... Hmm...

Instead, could there be any methods we can link between the degrading rating in the script (`distort_extent`) to a minimum mesh edge length of the final product? It could be used as a criterion compared with the ray resolution to calculate whether the distorted mesh with being captured in the adjacent ray.

mabelzhang, 2022-02-08:
Engineering-wise, it can probably implemented as a different mode - I want to keep the deformation script as generic as possible and not tweak it specifically for this sonar plugin. It would probably have to be iterative, and we'd define a min edge length plus some range tolerance. We'd distort it, check the min edge length stats, and if it isn't within the tolerance, re-distort, repeat until we are within tolerance. There can be situations where we never get to the edge length we want, in which case it'd return some error.

Before we go and implement that, let's check our mutual understanding. By edge length, you mean the polygon edges in the mesh, or something higher level? I have questions whether the polygon edge lengths can be reliable, because, using the cylinder as an example, you can see the vertical edges of the wall are very long - 2 m as the label says, whereas the edges on the top and bottom circles are 0.2 m. With that big a deviation, will getting the min edge length (0.2 m in this case) work for you? Depending on how big the object is wrt the sensor, and which part the sonar is looking at, that edge length might not be the length the sonar is looking at, right? (Say the sonar is looking at the wall of the cylinder that's 2 m, but I give you min edge length 0.2 m).

That kind of variation will always be there, depending on the object's geometry. We can subdivide, but as you pointed out, subdivide just cuts an edge in two, so the ratio between them will still be the same. (I just tested it by applying subdivision level 2, and I get 0.5 m and 0.05 m for the two lengths above).

woensug-choi, 2022-02-09:
Yes, but slightly different. I mean polygon edges that are **extended from the original mesh. This is the length of offset from the subdivided vertices.** And you are totally right about the ambiguity of using the min edge length to expect 'different' in the sonar image considering complex shapes of the object and relative angle from the sonar. And I understand that the `simple` method in the blender subdivision will keep the ratio of max and min edge lengths.

What I am trying to achieve is, while I am not completely bought to the idea of recognizing 'difference' in the sonar image since we do not know the use cases, to provide an example of how much they should distort the object using the `distort_extent` or whatsoever we might need to introduce additionally (i.e. min edge length).

The case I am imagining is for the object, let's say a cylinder, at a fixed distance from the sonar. When they are testing manipulation scenarios that require recognizing an object, something like measuring the length of the cylinder in the image (whichever orientation it is, maybe by using machine learning), if the object is degraded it will be shown as a different length in the real world.  To test their manipulation capability for the degraded objects in the simulation, they need to test both undegraded and degraded objects. We are trying to provide what value to use for the degradation script to generate one. The larger the better of course, but I am trying to provide min edge length as the minimum criterion to distort. Does this make sense?

woensug-choi, 2022-02-25:
Closing it. It will be tracked at https://github.com/Field-Robotics-Lab/nps_uw_multibeam_sonar/issues/39. Discussions may continue there.

closed 2022-02-25 by woensug-choi