Field-Robotics-Lab/dave #195: Add object models to ignition/fuel

bsb808, 2022-02-11:
Once we have the new object models integrated into the project for the next release - i.e., we have added the files and reviewed PRs such as #194 - then it would be good to upload these to the DAVE collection on ignition fuel.

Instructions from @mabelzhang 

YI a bit more info

You can upload newer versions of the same model, if you want to modify a previous upload.

There are some requirements for the models, so that they will work in Gazebo. There's very limited documentation here
[https://openrobotics.freshdesk.com/support/solutions/articles/43000011289-contributing-a-new-model](https://nam10.safelinks.protection.outlook.com/?url=https%3A%2F%2Fopenrobotics.freshdesk.com%2Fsupport%2Fsolutions%2Farticles%2F43000011289-contributing-a-new-model&data=04%7C01%7Cbbingham%40nps.edu%7C4528a981f2b74b395b3008d9ed930c21%7C6d936231a51740ea9199f7578963378e%7C0%7C0%7C637802038674180834%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=Lyij2X4QgkF5jVMIGv2dcy3SNRSRay2HxSVPu1VoujU%3D&reserved=0)

We package the models with top-level model.sdf and model.config files, put the meshes and textures in subfolders, etc. I don't know if it's Ryan or Duane on your end who handles that.

Let me know if any questions arise. We can probably improve our documentation if this turns out to be a pain.
Thanks!
Mabel

After creating an account on [app.ignitionrobotics.org](https://nam10.safelinks.protection.outlook.com/?url=http%3A%2F%2Fapp.ignitionrobotics.org%2F&data=04%7C01%7Cbbingham%40nps.edu%7C4528a981f2b74b395b3008d9ed930c21%7C6d936231a51740ea9199f7578963378e%7C0%7C0%7C637802038674180834%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=Oas%2BRQrYcUiACSyqxNRPqLvx9sg3XScdCcfDoWbkWDE%3D&reserved=0) (anyone can create an account), click the + button on the upper-right to upload a model or a world.

Per Cole, after uploading the model, on the model page, click this button (two overlapped squares of images) to add to a collection:

IgnitionFuel_uploadToCollection1.png

Then you can enter the name of a collection to look for it:

IgnitionFuel_uploadToCollection2.png

Hope that helps!
Mabel

On Fri, Feb 11, 2022, 12:09 Bingham, Brian (CIV) <[bbingham@nps.edu](mailto:bbingham@nps.edu)> wrote:
Hi Mabel,

Is there a way for Ryan to upload models to the DAVE collection at fuel - [https://app.ignitionrobotics.org/Cole/fuel/collections/DAVE](https://nam10.safelinks.protection.outlook.com/?url=https%3A%2F%2Fapp.ignitionrobotics.org%2FCole%2Ffuel%2Fcollections%2FDAVE&data=04%7C01%7Cbbingham%40nps.edu%7C4528a981f2b74b395b3008d9ed930c21%7C6d936231a51740ea9199f7578963378e%7C0%7C0%7C637802038674180834%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=7nIo%2Ft0EN9OOU9cY3p%2Bf%2BHLXDvQ5HEx5kIq7qUgFdPo%3D&reserved=0)

It would be really cool to have that collection be fairly complete.

All the best,
Brian

mabelzhang, 2022-02-12:
Additional tip:
Once a model is uploaded to Ignition Fuel, it can be included in the SDF file by `<uri>https://...</uri>`. The address is on the object page in the bibtex block where it says `url={...}`.

mabelzhang, 2022-02-17:
Tutorial for uploading models to Fuel https://ignitionrobotics.org/api/gazebo/6.5/meshtofuel.html
Models should be in a directory following the structure detailed in the tutorial.

quarkytale, 2022-03-22:
All set: [https://app.ignitionrobotics.org/OpenRobotics/fuel/collections/DAVE](https://app.ignitionrobotics.org/OpenRobotics/fuel/collections/DAVE)

mabelzhang, 2022-03-22:
I think we need to merge a PR replacing all the existing local references to models with the Fuel URLs before this can be closed :)