Field-Robotics-Lab/dave #208: GUI rendered dark in some systems

quarkytale, 2022-02-19:
The GUI is loading darker for some systems.

Wiki GUI:
[image]

Darker:
[image]

And really light:
[image]

Need to investigate what's causing this.

Notes from previous discussions:
- Possibly due to different driver versions, if the driver very out of date.
- No errors in OGRE logs

@WilliamLewww please add your findings too!

mabelzhang, 2022-02-19:
Originally discussion from https://github.com/Field-Robotics-Lab/dave/pull/186#discussion_r798897916

WilliamLewww, 2022-02-28:
I think that this issue has something to do with shadows. I was able to reproduce the same effect on two different machines where one would render everything lightly and one would render everything dark.

When digging into the shaders for the lightly rendered system, I was able to tinker with the shadow specific parts to get a similar result to the dark system. I'm thinking that the dark system are somehow rendering shadows over every surface while the light simulation doesn't render shadows at all.

[image]
[image]

bsb808, 2022-03-01:
Wow - that is a very subtle bug to track down - nice work!

mabelzhang, 2022-03-01:
I'm surprised if this is the first time it's been discovered though, if there's nothing unusually specific about the way lights are used in the DAVE worlds - are they?

I found this issue - is it related?
https://github.com/osrf/gazebo/issues/2686

There are a few shadow-related issues in osrf/gazebo
https://github.com/osrf/gazebo/issues?q=is%3Aissue+is%3Aopen+shadow
I'm just looking to see if there's already an existing ticket about it, otherwise the troubleshooting would go upstream.