Field-Robotics-Lab/dave #224: Moving  dave_sim.repos

bsb808, 2022-02-25:
At this point this is just to link to the comment in the merged PR so that it doesn't fall through the cracks.  @M1chaelM is probably best positioned to describe what needs to be done post 4.3.0 release.

See: https://github.com/Field-Robotics-Lab/dave/pull/199

> This looks good, and we're going to try it out. It sounds like removing the dave repos from dave_sim.repos would ultimately make sense also, though we will do that later just for logistical reasons relating to our upcoming release.

M1chaelM, 2022-02-25:
This is an optional optimization, so I was going to ask in our meeting whether we want to do it.

The situation is that when `vcs` was set up, the person who created the `dave_sim.repos` file inlcuded the dave repo in the file. This leads to a little bit of redundancy because you have to retrieve the dave repo to get the `dave_sim.repos` file in the first place. Then when you use `vcs` to get the rest of the repos for the first time, you have to ignore the fact that you downloaded the `dave` repo already. After the first time, it's very easy to use vcs to update all the repos at once, including the dave repo.

We've been living with this tiny bootstrapping requirement and using the `--skip-existing` flag to avoid getting an error the first time we run `vcs` to clone all the repositories. Since we are usually not wiping out our workspace and starting from scratch, this is not a big problem. 

Now that we are moving to a more sophisticated CI script that is going to rely on our `dave_sim.repos` file, having the dave repository in it makes the CI take a bit longer. The reason is that the CI script process is to automatically build and test the main repo (dave) and then build and test everything in the `dave_sim.repos` file. Since that file currently includes the dave repo itself, we end up testing it twice.

All of this is just a restatement of what our kind contributor pointed out when submitting the PR to add `industrial_ci`. There are two ways to avoid this redundancy and make the CI faster:

1. Remove the dave repo from `dave_sim.repos`. Our contributor says this is "preferred." If that's the case then it means we have probably not been using `vcs` in the best way, and there is some better workflow. This seems plausible but I'm not yet sure what that workflow is, so we need to learn a little more about `vcs`. So this change would affect our current workflow and installation instructions.
2. Keep `dave_sim.repos` the same, but create a new file, say `dave_ci.repos` that we use just for CI. This gets us the efficiency in the CI without need to update our own brains or processes. However, I agree with the contributor that this is not preferred. Having two versions of anything always incurs a synchronization cost and is very error prone, so I think there's a strong likelihood that somewhere down the line we'll update `dave_sim.repos` and forget to update `dave_ci.repos` and then have a CI workflow that doesn't test what we want it to test. So I'm hoping we can figure out the right way to use `vcs` and go with option 1.

M1chaelM, 2022-02-25:
Fixed the title since this issue is not about moving the `dave_sim.repos` file.

M1chaelM, 2022-02-25:
After some further testing, it looks like there is no good reason to keep the dave repo listed in the `dave_sim.repos` file. We still have the same install process but we no longer need the `--skip-existing` option in the initial import. After the initial import, `vcs` does not rely on the `.repos` file for things like pulls and updates. Instead, it automatically finds all the repositories under the present working directory by default, so it will include `dave` anyway.

In light of this, I'm going to make a simple PR to update the `repo` file.

bsb808, 2022-02-25:
Great description @M1chaelM - thank you.

woensug-choi, 2023-11-29:
Closing this since the PR is merged

closed 2023-11-29 by woensug-choi