Field-Robotics-Lab/dave #233: Rocker dep

romulogcerqueira, 2022-03-07:
Hi,

how can I install this dep? I can not find it on ubuntu sources...

I have tried to test the simulator using docker:

```pre
romulo@N8xxEP6:dockwater (cuda)$ ./run.bash noetic:latest
Using image <noetic:latest> to start container <noetic_runtime>
./run.bash: line 72: rocker: command not found

```

bsb808, 2022-03-07:
We've linked to the rocker install instructions in the wiki here: https://github.com/Field-Robotics-Lab/dockwater/wiki/Install-Dependencies

Also, based your previous work, I suspect you are interested in the sonar modeling.  The https://github.com/Field-Robotics-Lab/nps_uw_multibeam_sonar project continues to be active and the main author, @woensug-choi , is doing some very useful work to model both forward-looking and mapping sonar.

woensug-choi, 2022-03-08:
@romulogcerqueira Hi! Your work was very impressive and inspiring! If there's anything you need, tag me! I would love to be in service for your table.

romulogcerqueira, 2022-03-08:
Hi @bsb808 and @woensug-choi,

I was looking for underwater robotics simulators and I found your repository. I installed the rocker package and built the container successfully, but seems I need a joystick to run the simulator properly. I will read the documentation further.

I am also curious about sonar modeling, and it is very important to have more people working on that to benefit the whole maritime community. Congratulations on all the work done @woensug-choi, I will follow it from now.

closed 2023-10-31 by mabelzhang