Field-Robotics-Lab/dave #248: Movit_commander package.xml build error

woensug-choi, 2022-03-25:
```bash
 ~/uuv_ws  catkin_make
Base path: /home/woensug/uuv_ws
Source space: /home/woensug/uuv_ws/src
Build space: /home/woensug/uuv_ws/build
Devel space: /home/woensug/uuv_ws/devel
Install space: /home/woensug/uuv_ws/install
Creating symlink "/home/woensug/uuv_ws/src/CMakeLists.txt" pointing to "/opt/ros/noetic/share/catkin/cmake/toplevel.cmake"
Error(s) in package '/home/woensug/uuv_ws/src/dave/urdf/robots/rexrov_oberon7_moveit/package.xml':
Error(s):
- The manifest of package "rexrov_oberon7_moveit" (with format version 1) must not contain the following tags: exec_depend
- Either update to a newer format or replace <exec_depend> tags with <run_depend> tags.
```

```bash
 ~/uuv_ws  catkin build
Error: The file /home/woensug/uuv_ws/src/dave/urdf/robots/rexrov_oberon7_moveit/package.xml is an invalid package.xml file. See below for details:
Error(s):
- The manifest of package "rexrov_oberon7_moveit" (with format version 1) must not contain the following tags: exec_depend
```

Am I alone? Changing `exec_depend` to `run_depend` solved the problem.

@crvogt @Yadunund

Yadunund, 2022-03-25:
My apologies @woensug-choi this regression is caused by https://github.com/Field-Robotics-Lab/dave/issues/248.
I've opened https://github.com/Field-Robotics-Lab/dave/pull/250

mabelzhang, 2022-03-25:
I think the comment above means to tag #239 :)

closed 2022-03-25 by mabelzhang