Field-Robotics-Lab/dave #262: Plans to migrate Project DAVE to ROS2?

JSaunders97, 2022-11-21:
Hi, 

I was just wondering if there are plans to migrate Project DAVE to ROS2 in the future?

Thanks!

mabelzhang, 2023-06-26:
Hi, I don't have the authority to answer this in the absolute sense. But as far as I'm aware, the project has not received additional funding to go forward.

mabelzhang, 2023-07-08:
There are multiple projects from Open Robotics targeting ROS 2 and the [new Gazebo](https://gazebosim.org/), which has native plugins for marine features, rather than third-party plugins. Check out [LRAUV](https://github.com/osrf/lrauv/), [WEC](https://github.com/osrf/mbari_wec) for underwater, and [VRX](https://github.com/osrf/vrx) and [MBZIRC](https://github.com/osrf/mbzirc) for surface vehicles.

I will close this as the question has been answered. Feel free to reopen if there are related followup questions.

closed 2023-07-08 by mabelzhang