Field-Robotics-Lab/dave #266: UUV simulator is now deprecated

magnus-rovco, 2023-04-24:
What are the plans for DAVE now that UUV simulator is deprecated and archived?
Are there plan to take on the maintenance of UUV or move away from it?

mabelzhang, 2023-07-08:
As far as I heard, UUV simulator has long been out of actual maintenance. DAVE maintainers don't have relations with UUV sim maintainers. DAVE just builds on top of UUV sim.

DAVE itself is not under active development anymore.

I will note that both DAVE and UUV sim are based on Gazebo-classic, which will reach end of life in 2025.
There are multiple projects from Open Robotics targeting ROS 2 and the [new Gazebo](https://gazebosim.org/), which has native plugins for marine features, rather than third-party plugins. Check out [LRAUV](https://github.com/osrf/lrauv/), [WEC](https://github.com/osrf/mbari_wec) for underwater, and [VRX](https://github.com/osrf/vrx) and [MBZIRC](https://github.com/osrf/mbzirc) for surface vehicles.

mabelzhang, 2023-07-08:
Closing as question answered. Feel free to reopen with related questions.

closed 2023-07-08 by mabelzhang