Field-Robotics-Lab/dave #270: Where are the Virgil files that are used to new underwater vehicle tutorial?

whoi-engineer, 2023-05-31:
I've tried following the new underwater vehicle tutorial but I am hitting roadblocks. The tutorial would be easier to follow and then modify with ones own vehicle if the documents for the tutorial were actually on git. Is there something I'm missing?

mabelzhang, 2023-06-26:
> if the documents for the tutorial were actually on git

Are you referring to the tutorials for this repo? It is on Git: https://github.com/Field-Robotics-Lab/dave.doc (the GitHub repo name for GitHub IO pages are usually the last part of the URL)

mabelzhang, 2023-06-26:
Answering the title of the issue, the virgil files I believe are in a PR
https://github.com/Field-Robotics-Lab/dave/pull/244

It was one of the last PRs for the project, and the project has since been out of funding. The PR has gone through a long review as you can see. Feel free to give it a try.