Field-Robotics-Lab/dave #271: Can't get any of the tutorial robots moving

random-engineer89, 2023-06-02:
I've been playing around with the project Dave uuvs/auvs and while I am able to run most things, I have not yet been able to launch anything where the auv actually moves. The thrusters are able to spring but the robot itself isn't moving. 
I've tried multiple examples/tutorials from the project Dave home page, as well as from the rexrov2 and eca_a9 pages but none of the robots are moving. 

Examples of what I've tried:
Example 1:
    1. roslaunch uuv_gazebo_worlds empty_underwater_world.launch
    2. roslaunch rexrov2_description upload_rexrov2.launch
    3. roslaunch rexrov2_control start_sf_controller.launch
    4. roslaunch rexrov2_gazebo start_demo_pid_controller.launch
    5. roslaunch uuv_control_utils start_helical_trajectory.launch uuv_name:=rexrov2 n_turns:=2

Example 2:
    1. roslaunch dave_demo_launch dave_integrated_demo.launch
    2. roslaunch dave_nodes integrated_world_teleporter.launch

Example 3:
    1. roslaunch dave_robot_launch caracara.launch
    2. roslaunch dave_demo_launch dave_integrated_demo.launch

Please point me towards a simulation where the  auv actually moves in the environment.

mabelzhang, 2023-06-26:
Are you running a controller or some kind of teleop to move the robots?

To clarify, DAVE as a project was designed to provide world environments, vehicle models, and simulated sensors, etc. It is not a project where algorithms are developed. It provides a library for researchers and developers to develop and test their own algorithms easier, so that they don't have to reinvent the wheel and build the base environment.

For teleop, joystick mapping documentation is here:
https://field-robotics-lab.github.io/dave.doc/contents/manipulator_demos/Logitech-F310-Gamepad-Mapping/
Many of the manipulation tutorials have examples of how to use the joystick, e.g. this one:
https://field-robotics-lab.github.io/dave.doc/contents/manipulator_demos/Electrical-Plug-Mating-Plugin/

closed 2023-10-31 by mabelzhang

markusbuchholz, 2023-11-10:
@random-engineer89 

@mabelzhang  is correct. 

I noticed your post. If you need to run your robot, just check the topic in your environment.
If you have ``` \cmd_vel``` you can simply move your robot as follows:

```bash
catkin_create_pkg robot_control std_msgs rospy geometry_msgs
cd ~/catkin_ws/src/robot_control
cd ~/catkin_ws/src/robot_control
touch move_robot.py
chmod +x move_robot.py
cd ~/catkin_ws
catkin_make
source ~/catkin_ws/devel/setup.bash

```
Your script ```move_robot.py```  is:

```python
#!/usr/bin/env python
import rospy
from geometry_msgs.msg import Twist

def move_robot():
    rospy.init_node('uuv_robot_mover', anonymous=True)
    pub = rospy.Publisher('/your_robot/cmd_vel', Twist, queue_size=10)
    rate = rospy.Rate(20)  # 20 Hz

    while not rospy.is_shutdown():
        twist = Twist()
        twist.linear.x = 0.5  # set linear speed
        twist.angular.z = 0.0  # set angular speed
        pub.publish(twist)
        rate.sleep()

if __name__ == '__main__':
    try:
        move_robot()
    except rospy.ROSInterruptException:
        pass
```
Run your robot:

```bash
rosrun robot_control move_robot.py
```