Field-Robotics-Lab/dave #272: Build Dave | Error

Z8MAN8, 2023-06-21:
Hello, when I build a project based on the documentation, I encounter the following problem：
-- +++ processing catkin package: 'nps_uw_multibeam_sonar'
-- ==> add_subdirectory(nps_uw_multibeam_sonar)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- Could NOT find acoustic_msgs (missing: acoustic_msgs_DIR)
-- Could not find the required component 'acoustic_msgs'. The following CMake error indicates that you either need to install the package with the same name or change your environment so that it can be found.
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "acoustic_msgs"
  with any of the following names:

    acoustic_msgsConfig.cmake
    acoustic_msgs-config.cmake

  Add the installation prefix of "acoustic_msgs" to CMAKE_PREFIX_PATH or set
  "acoustic_msgs_DIR" to a directory containing one of the above files.  If
  "acoustic_msgs" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  nps_uw_multibeam_sonar/CMakeLists.txt:17 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/csc/uuv_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/csc/uuv_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed

I can't solve this problem myself, can you help me？Thank you

mabelzhang, 2023-06-26:
Hi, this is an upstream error. We encountered this in ROS 2 Humble in another project. Our solution is to revert to an earlier commit to bypass that error https://github.com/osrf/mbari_wec/pull/51/files

Are you using ROS 2? We traced the following for why it didn't work for Humble. For other distros, or ROS 1, it might be something similar but not exactly the same.

The cause is that [`ros_gz`](https://github.com/gazebosim/ros_gz) has a new dependency on [`actuator_msgs`](https://github.com/rudislabs/actuator_msgs), which had been released into Humble https://github.com/ros/rosdistro/pulls?q=is%3Apr+actuator_msgs+ But at the time we encountered the error, Humble didn't have a sync yet, so there was no binary package like ros-humble-actuator-msgs on Ubuntu 22.04.

There has been some releases of ros_gz on rosdistro since we saw the problem on May 2. e.g. this one into Humble on May 23 https://github.com/ros/rosdistro/pull/37346

A Humble sync has come out on May 30 https://discourse.ros.org/t/preparing-for-humble-sync-2023-05-30/31622/2

So maybe it's better now? I haven't checked recently.

mabelzhang, 2023-06-26:
Ah, sorry, in the above comment, I was confused by the name `acoustic_msgs`. There is an upstream message called acoustic_msgs, but the one used in DAVE is a completely different package from APL https://github.com/Field-Robotics-Lab/nps_uw_multibeam_sonar/blob/50c34356513d6d0827eb5ad6f5cc691caf06d7ab/package.xml#L23

This PR should fix your error https://github.com/Field-Robotics-Lab/dave/pull/269
Let us know if it fixes it for you, so that we can merge that PR and close this issue.
Thanks!

CCLings, 2023-08-15:
In the folder `uuv_ws/src`, this project cloned  `acoustic_msgs` package from https://github.com/apl-ocean-engineering/marine_msgs/tree/main/marine_acoustic_msgs to the folder `hydrographic_msgs`. But unfortunately, the package has been updated, NOT compatible. If you just want to run codes, the simplest way is the rollback of the downloaded package.
```
cd uuv_ws/src/hydrographic_msgs
git checkout 5344060071d0a72857a9efb558cc1f2d7c0052c6
```

mabelzhang, 2023-10-31:
Yeah, #269 updates that dependency.
I haven't had time to test out that PR, though I believe it most likely works because it comes from the authors of hydrographic_msgs. If anyone has time to test it and post an approval there, and we have multiple approvals, I'll merge it.

huangbintao188, 2023-11-09:
Could you tell me how to solve this? thank you.

woensug-choi, 2023-11-29:
@huangbintao188  working on this. I will ping you back when I get the update uploaded.