Field-Robotics-Lab/dave #276: Support for ROS2 humble?

kulkarni-raunak, 2023-10-30:
Hello Field-Robotics-Lab community,

Is there a support for this simulation to run with `ROS2 humble` , `Ubuntu Jammy 22.04` and `Gazebo garden`?

mabelzhang, 2023-10-31:
Not currently, but we'd love to see community efforts to port it.

I'm going to make a motion at the next ROS Maritime Working Group (next Tue, Nov 7, announced [here](https://discourse.ros.org/t/maritime-working-group-meeting-nov-2023-technical-working-meeting/34327)) for the community to pick up some work in this repo, and porting will be one of them.

I've been given some good ideas for how to start that, beginning from simple things like SDF models. Plugins will be the hardest. I'll create some issue tickets and add them to the WG [kanban](https://github.com/orgs/ros-maritime/projects/1) board to make them visible to the WG community.

woensug-choi, 2023-11-29:
@mabelzhang I want to give it a try.... not sure I can finish it. Which version pair of ROS and Ignition should I aim for? 
 `Ubuntu Jammy 22.04`, `ROS2 humble` and `Gazebo Fortress`?

mabelzhang, 2023-11-29:
Sigh... that's a good question. The latest is Gazebo Ionic, which is targeting ROS 2 Jazzy.

The official recommendation is in the table here https://gazebosim.org/docs/harmonic/ros_installation

Ionic and Jazzy pairing is not on there yet, but that's the plan.

Fortress is way too old. I do not recommend it, even if that is the recommended green check. We were able to get ROS 2 Humble and Gazebo Garden to work just fine for [MBARI WEC](https://github.com/osrf/mbari_wec), so I would do at least that, or newer. All the newer underwater features we added from WEC, like added mass etc., are in Garden.

mabelzhang, 2023-11-29:
I can open a more formal ticket for it. Evan Palmer was looking to port to ROS 2 too.

woensug-choi, 2023-11-30:
So... 

1. ROS Jazzy (LTS ~ 2029) and Gazebo Harmonic (LTS ~ 2028)
2. ROS Humble (LTS ~2027) and Gazebo Garden (~2024)
3. ROS Iron (~2024) and Gazebo Harmonic (LTS ~2028)

so many choices... After listing the options, I see Jazzy Harmonic is a best choice to put efforts..

mabelzhang, 2023-12-02:
That sounds reasonable for both to be LTS.

For installation, one would want to follow the instructions on this [page](https://gazebosim.org/docs/harmonic/ros_installation), under "Gazebo Harmonic can be used with ROS 2 Rolling but ros_gz will need to be compiled from source." There's also a step to install rosdep rules for Harmonic.

mabelzhang, 2023-12-06:
Yeah, Jazzy is to be released May 2024. Right now Rolling is what will become Jazzy.

So, if we want to do Jazzy+Harmonic pairing, that will work once Jazzy is released. Before Jazzy is released, it unfortunately will need to be Rolling built from source.

Iron + Harmonic would be the next-best pairing. Both are currently the latest released versions, so they can be installed from binaries. When Jazzy is released, then the code can be updated. Hopefully from Iron to Jazzy will not be a lot of work... I am not familiar with what's on the ROS 2 roadmap to say for sure. I've asked internally and will update here.

Oh no that file is all the packages for ROS 2. It's massive. You wouldn't want to look at that.

You only need to port packages within DAVE, so packages in this repo.
Note that we're talking about 2 different migrations. ROS 1 to ROS 2. Gazebo-classic to new Gazebo. I would do them one at a time. (As in, you can use ROS 1 with new Gazebo, and ROS 2 with Gazebo-classic. So I wouldn't change the code for both ROS 2 and new Gazebo at the same time, it could become very hairy because you'll also need to think about gazebo_ros_pkgs to gz_ros_pkgs.)

For ROS migration, I suspect it's mainly the ROS plugins. https://github.com/Field-Robotics-Lab/dave/tree/master/gazebo/dave_gazebo_ros_plugins
package.xml should be straight forward.
Launch files should keep using the XML format in ROS 2. Do not convert to Python launch files - that's more of a pain.
We have tutorials for migrating ROS 1 to ROS 2 https://docs.ros.org/en/iron/How-To-Guides/Migrating-from-ROS1.html

For Gazebo migration, that's models and Gazebo plugins. Those tutorials are here https://gazebosim.org/api/sim/8/tutorials.html

Lastly, for combining the two, there's tutorials for ROS 2 + new Gazebo integration here https://gazebosim.org/docs/harmonic/tutorials#ros-integration

There's also URDFs.
This section explains some latest updates to URDF/SDF: "The main pain point of using existing simulation assets with ROS 2 tools was updating URDF files into a Gazebo-readable format. This is no longer required. If you are maintaining a URDF and an SDF file in a project, you can now drop the URDF and just use the SDF for both ROS and Gazebo."
https://gazebosim.org/docs/harmonic/ros2_interop#maintaining-a-single-robot-description-format

mabelzhang, 2023-12-06:
Yeah. Let me try to put an initial meta-checklist ticket together by end of the week. I probably won't get to it today. Hopefully by Friday.

mabelzhang, 2023-12-07:
Jazzy will be paired with Ionic (as mentioned in https://github.com/Field-Robotics-Lab/dave/issues/276#issuecomment-1831930143).

Right now, since Jazzy isn't released yet, if DAVE targets that, DAVE developers will need to build ROS from source, which can be an overhead. Users who are not developers probably won't want to build ROS from source. On the flip side, given that migration will take significant time, maybe by the time migration is done, Jazzy will have been released.

Whichever pairing we do, we will need to provide a Dockerfile on `dockwater`, which hopefully lightens the burden of compiling ROS.

> Instead of deleting the models altogether, it might be nice to have a separate repo (or even continue using this one) for new users to contribute models toward.

I would recommend new users to contribute models directly to Fuel, as opposed to this repo, because this repo will have the same problem of not being maintained. The community has been contributing models to Fuel for other projects, like SubT and our other client projects. In fact, we already started doing that for new models we created for DAVE - you can see the collection here https://app.gazebosim.org/OpenRobotics/fuel/collections/DAVE We just didn't have time to upload the rest. Now would be a good time to migrate the rest to Fuel.

mabelzhang, 2023-12-08:
I've created the promised meta-ticket in #283.
Please move discussions there. I'm going to close this ticket so that threads are visible in one place in the new ticket.

closed 2023-12-08 by mabelzhang

mabelzhang, 2023-12-22:
Sorry folks, I misremembered in my answers above. Gazebo Harmonic will be paired with ROS Jazzy. Both are LTS.
We should probably put that somewhere in the documentations...
That should make this choice a lot easier.

woensug-choi, 2023-12-22:
> So... 
> 
> 1. ROS Jazzy (LTS ~ 2029) and Gazebo Harmonic (LTS ~ 2028)
> 2. ROS Humble (LTS ~2027) and Gazebo Garden (~2024)
> 3. ROS Iron (~2024) and Gazebo Harmonic (LTS ~2028)
> 
> so many choices... After listing the options, I see Jazzy Harmonic is a best choice to put efforts.. 

It is in the documentation! Above was my summary of it

mabelzhang, 2023-12-22:
Are you sure? I didn't find any mention of Jazzy on this page https://gazebosim.org/docs/harmonic/ros_installation If it is official pairing, the table should list Jazzy (Rolling) + Harmonic as a green check. Right now it's not.

mabelzhang, 2024-01-05:
Harmonic + Rolling pairing has been added! https://github.com/gazebosim/docs/pull/420#pullrequestreview-1804975060

mabelzhang, 2024-01-17:
And more explicitly, Harmonic + Jazzy pairing https://github.com/gazebosim/docs/pull/421